viveris/uMTP-Responder #2: Logo proposal

michaelizer, 2018-04-17:
Hello, I just made a logo for your project, hope you liked it.
[image]
[image]
Let me know if you liked it and which variation and I'll upload the files on a PR. 
Please tell me if you want to modify something,

jfdelnero, 2018-04-18:
First Thanks for these nice logos !

In fact we are a bit concerned about the USB logo. This one is trademarked and this may eventually cause some legal issues : 

http://www.usb.org/developers/logo_license/

See page 5:
http://www.usb.org/developers/logo_license/USB-IF_Logo_Usage_Guidelines_FINAL_101917.pdf

michaelizer, 2018-04-19:
So sorry, totally forgot about that. Check these out 
[image]
[image]

jfdelnero, 2018-05-23:
Yes this logo look fine !

closed 2018-05-24 by Jean--Michel