viveris/uMTP-Responder #5: libcomposite support

jackmitch, 2018-05-25:
First off, fantastic project; I've deployed this sucessfully on an Allwinner based platform without issue so great work!

What is the current status of the libcomposite support and if nothing has been started yet, how complex is the transition?

shenghaoyang, 2018-05-25:
Willing to work on this too, enjoying it on a Pi right now. Pretty funny to see it emulating so many devices.

jfdelnero, 2018-05-28:
@jackmitch Thanks for the feedback. BTW which board it is exactly ? Can you share the USB endpoints settings to use ?

About libcomposite : From my analysis, it appears that the right interface to use is FunctionFS since uMTP-Responder is a pure user-land "driver". From what i have read, libcomposite is a kernel library and FunctionFS allows to add/emulate an usb function from the user-land. Both libcomposite and FunctionFS should be able to work together to make a composite usb device with the kernel gadgets and the user uMTP-Responder driver/daemon.

The transition shouldn't be complex. We plan to keep the GadgetFS support and add the FunctionFS support. The interface selection will be done at compile time.

jfdelnero, 2018-12-05:
libcomposite/FunctionFS is now supported by uMTP-Responder v0.9.2

closed 2018-12-05 by jfdelnero