viveris/uMTP-Responder #9: dwc2

vanti, 2018-09-29:
Hi, I have built umtprd on beaglebone black running their default Debian image (Debian 9.5 2018-08-30 4GB SD IoT), and put it under /usr/bin. I have gadgetfs, but got an error when running umtprd.sh:
modprobe: FATAL: Module dwc2 not found in directory /lib/modules/4.14.67-ti-r73

Looks like the script is trying to invoke modprobe on dwc2. What is the latter - is it necessary and what should I do to get it?

vanti, 2018-09-29:
Thanks. I removed that line, and now I am getting a different error from 'mount -t gadgetfs gadgetfs /dev/gadget':
mount: unknown filesystem type 'gadgetfs'

'modprobe gadgetfs' seems to work fine, and I got the following in dmesg:
[  133.545277] gadgetfs: USB Gadget filesystem, version 24 Aug 2004

Any idea what else I need to do?

jfdelnero, 2018-09-29:
This probably means that the usb device driver is not loaded. Check if you have an usb device driver ko module available. If not you may have to build it from the kernel sources.

vanti, 2018-09-29:
It seems I have the usb driver loaded:

debian@beaglebone:~/uMTP/uMTP-Responder/conf$ lsmod
Module                  Size  Used by
gadgetfs               24576  0
pruss_soc_bus          16384  0
evdev                  24576  1
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
8021q                  32768  0
garp                   16384  1 8021q
mrp                    20480  1 8021q
stp                    16384  1 garp
llc                    16384  2 garp,stp
usb_f_mass_storage     53248  2
usb_f_acm              16384  2
u_serial               20480  3 usb_f_acm
usb_f_ecm              20480  2
usb_f_rndis            32768  4
u_ether                20480  2 usb_f_ecm,usb_f_rndis
libcomposite           65536  18 usb_f_ecm,usb_f_acm,usb_f_mass_storage,usb_f_rn                                                                                                                                                             dis
iptable_nat            16384  0
nf_conntrack_ipv4      20480  1
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
nf_nat_ipv4            16384  1 iptable_nat
nf_nat                 32768  1 nf_nat_ipv4
nf_conntrack          143360  3 nf_conntrack_ipv4,nf_nat_ipv4,nf_nat
iptable_mangle         16384  0
iptable_filter         16384  0
spidev                 20480  0
pru_rproc              28672  0
pruss                  16384  1 pru_rproc
pruss_intc             16384  1 pru_rproc
ip_tables              24576  3 iptable_mangle,iptable_filter,iptable_nat
x_tables               36864  3 iptable_mangle,ip_tables,iptable_filter

Which image did you use for validation on beaglebone black? Mine is the default one (Debian 9.5 2018-08-30 4GB SD IoT) downloaded from their website: https://beagleboard.org/latest-images

jfdelnero, 2018-09-29:
Ok i see : You have the libcomposite layer loaded conflicting with the gadgetfs layer.
gadgetfs can't be used with libcomposite loaded.

2 solutions :  
- unload all libcomposite drivers and start gadgetfs.
or
- try to use umtprd in FunctionFS/libcomposite mode. This mode still in development.
See the umtprd-ffs.sh and umtprd.conf for the ffs mode setting example.

vanti, 2018-10-02:
Thanks. I decided to try the FunctionFS mode. I uncommented these lines in umtprd.conf:

```
usb_functionfs_mode 0x1

usb_dev_path   "/dev/umtp/ep0"
usb_epin_path  "/dev/umtp/ep1"
usb_epout_path "/dev/umtp/ep2"
usb_epint_path "/dev/umtp/ep3"

usb_max_packet_size 0x200
```

Then I ran the code in umtprd-ffs.sh. I had to change the executable permission on the files for 'echo' to work. Other then that, the script ran fine for the most part but it gives an error on the last line:

echo 300000.gadget > cfg/usb_gadget/g1/UDC
echo: write error: No such device

Any idea what went wrong? Also I noticed the line that is invoking 'umtprd' is commented out in the file. Was this intentional?

UHU-ger, 2018-10-19:
I got it running on the BeagleBoneBlack (with disabling libcomposite)
(Using BeagleBoard.org Debian Image 2018-10-07: bone-debian-9.5-lxqt-armhf-2018-10-07-4gb.img)
- with FunctionFS I got the same Problems you stated

How i got rid of libcomposite:

in the file `opt/scripts/boot/am335x_evm.sh`
i added the # to the folling lines in `use_libcomposite () { ` (possibly not the best way, but it worked)
```
#        echo "${log} modprobe libcomposite"
#        modprobe libcomposite || true
#        if [ -d /sys/module/libcomposite ] ; then
#                run_libcomposite
#        else
#                if [ -f /sbin/depmod ] ; then
#                        /sbin/depmod -a
#                fi
#                echo "${log} ERROR: [libcomposite didn't load]"
#        fi
```
after a restart `lsmod` shows libcomposite is gone

These are the commands i used to download and setup uMTP-Responder
```
cd ~
wget https://github.com/viveris/uMTP-Responder/archive/master.zip
unzip master.zip

cd uMTP-Responder-master
make
sudo cp umtprd /usr/bin/umtprd

cd conf/
sudo mkdir /etc/umtprd
sudo cp umtprd.conf /etc/umtprd/umtprd.conf
sudo nano /etc/umtprd/umtprd.conf
```
I added the # to the `-- GadgetFS Mode : Raspberry PI Zero / Raspberry PI Zero W Boards --` options 
and added
```
########################################################################
#
# -- GadgetFS Mode : BeagleBone Black --
#
########################################################################

usb_dev_path   "/dev/gadget/musb-hdrc"
usb_epin_path  "/dev/gadget/ep1in"
usb_epout_path "/dev/gadget/ep2out"
usb_epint_path "/dev/gadget/ep3in"

# Max USB packet size

usb_max_packet_size 0x200
```
at the end of the File
(you can instead change the `usb_dev_path` from the Pie enty to `"/dev/gadget/musb-hdrc"`

To start uMTP-Responder:
```
chmod +x ~/uMTP-Responder-master/conf/umtprd.sh
sudo ~/uMTP-Responder-master/conf/umtprd.sh
```

I hope it helps you (and others)
I would have been happy to find a Guide like this.

closed 2024-07-08 by jfdelnero