viveris/uMTP-Responder #10: Bug: Canceling a copy operation (Windows 10) will result in a crash (using BeagleBoneBlack - gadgetfs)

UHU-ger, 2018-10-19:
I mentioned this Bug while checking the speed of the connection.
I use the actual source-code from the master Branch

I copied a large File (170Mb) to test the speed of umtprd.
After about 10 seconds i canceled (as it had copied approx 50%).
Now the MTP-Device disappears. (not showing again, even with usb-cable out and in)
in linux screen nothing happened
i quit umtprd with CTRL-C (no output about the error) 
the already copied part of the file is still there
Can restart umtprd without problems.

jfdelnero, 2018-10-23:
The transfer cancellation mechanism is not yet implemented. working on this.
From the tests i have done, the mtp responder is not crashed, but you need to wait some seconds for the reconnection.

jfdelnero, 2019-09-30:
The transfer cancellation is now implemented and working on Linux and Windows 7 / 10.

closed 2019-09-30 by jfdelnero