viveris/uMTP-Responder #11: Changes made in the Filesystem (from within the device running umtprd) not updated

UHU-ger, 2018-10-19:
running master Branch on BeagleBoneBlack.

If i create/copy/delete something on a connected Windows 10 Computer the changes are visible from the Beagle. If i change something from the Beagle (create/rename/delete) the changes are not visible on the Windows 10 Device.

(I wanted to test to take a File copied from Windows to one "storage entry point" and move it with the linux running on the device (and later change something in) it to another "storage entry Point".)

P.S. I don't want to criticize, i am happy to have found a working mtp. I just want to let you know about the bugs i find while testing.

jfdelnero, 2018-10-23:
Unlike Linux, Windows 10 is relying on the event channel to refresh the folder if a file is added from the responder. This still need to be implemented.

gregoiregentil, 2018-12-01:
I also would like to see this feature too. I guess that you would need a separate thread in the daemon which would do an inotify watch for each folder. Not super complex but a pain to develop and handle.

Another (temporary) solution would be to have the ability to rebuild the whole database upon the reception of a SIGUSR1 signal (no d-bus, no dependency...). Is there a way in the code to easily rebuild the database?

jfdelnero, 2018-12-01:
All is almost there to update the database. But with Windows you also need to send some event messages on the third endpoint. 
Anyway this feature will be implemented for sure.

jfdelnero, 2018-12-22:
@gregoiregentil  @UHU-ger : The event support is now implemented on the trunk uMTPrd repository. Please have a try and let me know if you have any problem.

Notes : On Windows 10 the explorer doesn't appear to be refreshed automatically. The event is correctly received by Windows but F5 is needed to see the changes on the explorer. (This currently seems to be the case with any MTP device ! - tested also with a Galaxy J3...)

jfdelnero, 2019-01-10:
umtprd-0.9.7 with MTP events support is now released.

https://github.com/viveris/uMTP-Responder/releases/tag/umtprd-0.9.7

closed 2019-01-10 by jfdelnero