viveris/uMTP-Responder #12: kernel crash with Samsung Artik710

gregoiregentil, 2018-11-24:
I'm trying to make work uMPT-Responder on Samsung Artik710 via the USB OTG port.

I have "CONFIG_USB_GADGETFS=m" in the kernel 4.4:
https://github.com/SamsungARTIK/linux-artik/tree/A710_os_2.2.0

root@:~/uMTP-Responder# insmod gadgetfs.ko
root@:~/uMTP-Responder# mkdir /dev/gadget
root@:~/uMTP-Responder# mount -t gadgetfs gadgetfs /dev/gadget
root@:~/uMTP-Responder# ls /dev/gadget/
c0040000.dwc2otg

Note that there is no ep* in /dev/gadget

root@:~/uMTP-Responder# ./umtprd 
[Info] uMTP Responder
[Info] Version: v0.8.6 compiled the Nov 24 2018@04:56:45
[Info] (c) 2018 Viveris Technologies
[Info] Add storage root folder - Root Path: /
[Info] Add storage home folder - Root Path: /home/ai/inout/
[Info] USB Device path : /dev/gadget/c0040000.dwc2otg
[Info] USB In End point path : /dev/gadget/ep1
[Info] USB Out End point path : /dev/gadget/ep2
[Info] USB Event End point path : /dev/gadget/ep3
[Info] USB Max packet size : 0x200 bytes
[Info] Manufacturer string : Viveris Technologies
[Info] Product string : The Viveris Product !
[Info] Serial string : 01234567
[Info] Interface string : MTP
[Info] USB Vendor ID : 0x1D6B
[Info] USB Product ID : 0x0100
[Info] USB class ID : 0x06
[Info] USB subclass ID : 0x01
[Info] USB Protocol ID : 0x01
[Info] USB Device version : 0x3008
[Info] USB GadgetFS Mode
[Info] Wait for connection : 0
[Info] Loop on disconnect : 0

[   68.616000] DEBUG_LOCKS_WARN_ON(val > preempt_count())
[   68.616000] ------------[ cut here ]------------
[   68.616000] WARNING: at kernel/sched/core.c:2985
[   68.616000] Modules linked in: gadgetfs dhd cfg80211
[   68.616000] 
[   68.616000] CPU: 0 PID: 1197 Comm: umtprd Not tainted 4.4.71-00005-g2fc62ed-dirty #208
[   68.616000] task: ffffffc023bd8c00 ti: ffffffc020c98000 task.ti: ffffffc020c98000
[   68.616000] PC is at preempt_count_sub+0x60/0xd0
[   68.616000] LR is at preempt_count_sub+0x60/0xd0
[   68.616000] pc : [<ffffffc0000c0cb8>] lr : [<ffffffc0000c0cb8>] pstate: 80000145
[   68.616000] sp : ffffffc020c9bd20
[   68.616000] x29: ffffffc020c9bd20 x28: ffffffc020c98000 
[   68.616000] x27: ffffffc00082e000 x26: 0000000000000040 
[   68.616000] x25: 000000000000011d x24: 0000000000000015 
[   68.616000] x23: 0000000000000000 x22: 0000000000000004 
[   68.616000] x21: ffffffc020ce6e00 x20: ffffffc020ce6e00 
[   68.616000] x19: 0000000000000001 x18: 0000000000096640 
[   68.616000] x17: 0000007fb6f73930 x16: ffffffc0001df4b4 
[   68.616000] x15: 000000000000000a x14: ffffffc080d9029f 
[   68.616000] x13: 0000000000000005 x12: 0000000000000030 
[   68.616000] x11: ffffffc000d902ad x10: 00000000fffffffe 
[   68.616000] x9 : 0000000005f5e0ff x8 : 706d65657270203e 
[   68.616000] x7 : 206c6176284e4f5f x6 : ffffffc000d902d9 
[   68.616000] x5 : ffffffc00086df20 x4 : 0000000000000000 
[   68.616000] x3 : ffffffc00044956c x2 : 0000000000000000 
[   68.616000] x1 : 0000000000000000 x0 : 000000000000002a 
[   68.616000] 
[   68.616000] ---[ end trace 4d5f9b115a8c362a ]---
[   68.616000] Call trace:
[   68.616000] [<ffffffc0000c0cb8>] preempt_count_sub+0x60/0xd0
[   68.616000] [<ffffffc000826060>] _raw_spin_unlock_irq+0x30/0x58
[   68.616000] [<ffffffbffc214830>] dev_config+0x1a0/0x3a8 [gadgetfs]
[   68.616000] [<ffffffc0001de370>] __vfs_write+0x48/0xe8
[   68.616000] [<ffffffc0001dec44>] vfs_write+0xbc/0x158
[   68.616000] [<ffffffc0001df508>] SyS_write+0x54/0x98
[   68.616000] [<ffffffc000084630>] el0_svc_naked+0x24/0x28

Any idea what's wrong? I guess it's more a kernel bug.

gregoiregentil, 2018-11-25:
After some tweaks in kernel, config files and code, I have (eventually) managed to see the MTP drive using functionfs mode on an Ubuntu PC host.

Nevertheless, when I open the MTP drive, the uMTP daemon crashes:

[Error] incoming_packet : Read Error (-1)!
[Info] uMTP Responder : Disconnected

Has uMTP been tested in functionfs mode? There is a comment that functionfs should be selected during compilation in one of the issues but there is no relevant define anywhere.

jfdelnero, 2018-11-25:
This was tested but the functionfs support is quite young...

Can you share the full logs and your functionfs settings ?

gregoiregentil, 2018-11-26:
The files are:

loop_on_disconnect 0

storage "/home"  "home folder"

manufacturer "Viveris Technologies"
product "The Viveris Product !"
serial "01234567"

interface "MTP"

usb_vendor_id  0x1D6B
usb_product_id 0x0100
usb_class 0x6
usb_subclass 0x1
usb_protocol 0x1

usb_dev_version 0x3008

usb_functionfs_mode 0x1

usb_dev_path   "/dev/umtp/ep0"
usb_epin_path  "/dev/umtp/ep1"
usb_epout_path "/dev/umtp/ep2"
usb_epint_path "/dev/umtp/ep3"

usb_max_packet_size 0x200

#!/bin/sh

mkdir /tmp/config
mount -t configfs none /tmp/config
PATHg1=/tmp/config/usb_gadget/g1
mkdir $PATHg1
cd $PATHg1
mkdir $PATHg1/strings/0x409
mkdir $PATHg1/configs/c.1
mkdir $PATHg1/configs/c.1/strings/0x409
echo "Viveris Technologies" > $PATHg1/strings/0x409/manufacturer
echo "The Viveris Product !" > $PATHg1/strings/0x409/product
echo "01234567" > $PATHg1/strings/0x409/serialnumber
echo 0x1 > $PATHg1/bDeviceProtocol
echo 0x1 > $PATHg1/bDeviceSubClass
echo 0x6 > $PATHg1/bDeviceClass
echo "0x1D6B" > $PATHg1/idVendor
echo "0x0100" > $PATHg1/idProduct
echo "Conf 1" > $PATHg1/configs/c.1/strings/0x409/configuration
echo 120 > $PATHg1/configs/c.1/MaxPower
mkdir $PATHg1/functions/ffs.mtp
ln -s functions/ffs.mtp $PATHg1/configs/c.1
mkdir /dev/umtp
mount -t functionfs mtp /dev/umtp
/home/ai/mtp-server/uMTP-Responder/umtprd &
sleep 1
echo "`ls /sys/class/udc/`" > $PATHg1/UDC

On Linux PC, nothing appears. On Windows, the device appears in "Device Manager" as a MTP device but nothing is mounted. The logs are:

[Info] uMTP Responder
[Info] Version: v0.8.6 compiled the Nov 26 2018@20:43:06
[Info] (c) 2018 Viveris Technologies
[Info] Add storage home folder - Root Path: /home
[Info] USB Device path : /dev/umtp/ep0
[Info] USB In End point path : /dev/umtp/ep1
[Info] USB Out End point path : /dev/umtp/ep2
[Info] USB Event End point path : /dev/umtp/ep3
[Info] USB Max packet size : 0x200 bytes
[Info] Manufacturer string : Viveris Technologies
[Info] Product string : The Viveris Product !
[Info] Serial string : 01234567
[Info] Interface string : MTP
[Info] USB Vendor ID : 0x1D6B
[Info] USB Product ID : 0x0100
[Info] USB class ID : 0x06
[Info] USB subclass ID : 0x01
[Info] USB Protocol ID : 0x01
[Info] USB Device version : 0x3008
[Info] USB FunctionFS Mode
[Info] Wait for connection : 0
[Info] Loop on disconnect : 0
[Error] incoming_packet : Read Error (-1)!
[Info] uMTP Responder : Disconnected

gregoiregentil, 2018-11-27:
I'm no different and your config causes the same problem.

gregoiregentil, 2018-11-27:
Also, I think that your script has a typo. It's : mount -t functionfs mtp /dev/ffs-umtp

jfdelnero, 2018-11-27:
is umtp running as root ?

gregoiregentil, 2018-11-27:
yes

jfdelnero, 2018-11-27:
and also : is the usb port already connected when you start uMTP ?

gregoiregentil, 2018-11-27:
yes

jfdelnero, 2018-11-27:
The present Linux USB ID (0x1D6B:0x0100) are working perfectly on Windows 7/10 and Linux. libmtp shouldn't be modified.

Btw your kernel version is quite old (4.4). Is it an particular reason to keep this version ? Is it an Android patched kernel ?

Can you check that the others functionfs usb gadget are working ? For example just enable the acm gadget and disable the umtp gadget in the startup script : https://github.com/viveris/uMTP-Responder/blob/master/conf/umtprd-ffs.sh

gregoiregentil, 2018-11-27:
Yes, I have the serial, mass storage working well both in gadgetfs and functionfs mode. Also, I managed to have the MTP working well on Linux with the Android mtp patch and the Ubuntu phablet mtp-server (which has the BOOST dependency) but it's not working on Windows. The os_desc is not sent and I don't know why. For Linux, I need to have the Nexus 5X USB ID otherwise the drive doesn't appear on the Linux desktop.

I have tried your exact conf and script files. Windows shows it as a "MTP USB Device" in the device manager but nothing happens. On Linux, nothing happens also except if I put the Nexus 5X USB ID. Then it appears on the desktop but there is the error reported above as soon as I try to open the drive.

This is the Artik kernel 4.4.x. It has a lot of Samsung stuff in it so it's impossible to upgrade it myself.

jfdelnero, 2018-11-27:
Please add the DEBUG definition (-DDEBUG) to the CFLAGS into the makefile and rebuild umtprd. Then post the full logs here.

gregoiregentil, 2018-11-27:
I guess the next step is going to add debug info in the kernel? Note that I have recently added this patch: https://lore.kernel.org/patchwork/patch/590018/ (as I was adding the android MTP patch series and this was part of).

jfdelnero, 2018-11-27:
```
[Debug] Init end point /dev/ffs-umtp/ep1 (0)
[Debug] init_ep (0):
[Debug]
01 00 00 00 07 05 81 02 00 02 00 07 05 81 02 00
02 00
[Debug] Write error -1 (Cannot send after transport endpoint shutdown)
```

Ok interesting. It appears that this End point configuration is rejected for some reasons. Now we need to find why it is not accepted by your system. is it an USB3 port ?

(btw i have changed this debug message to error on the repository.)

gregoiregentil, 2018-11-27:
No. USB 2.0 OTG. Again, it's working (at least on Linux) with the mtp Android patch so the hardware "can" do it. But with the mtp Android patch, the path is different as it goes to configfs-mtp rather than configs-fs.

gregoiregentil, 2018-11-27:
Here is also the kernel log of the working scenario with f_mtp.c in the kernel and the device plugged to a Linux PC and running the Ubuntu phablet "mtp-server":

[  267.296000] configfs-gadget gadget: mtp_function_bind dev: ffffffc023468e00
[  267.296000] configfs-gadget gadget: create_bulk_endpoints dev: ffffffc023468e00
[  267.296000] configfs-gadget gadget: usb_ep_autoconfig for ep_in got ep1in
[  267.296000] configfs-gadget gadget: usb_ep_autoconfig for mtp ep_out got ep1out
[  267.296000] configfs-gadget gadget: usb_ep_autoconfig for mtp ep_intr got ep2in
[  267.296000] configfs-gadget gadget: dual speed mtp: IN/ep1in, OUT/ep1out
[  267.296000] dwc2 c0040000.dwc2otg: bound driver configfs-gadget
[  267.364000] mtp_open
[  267.372000] configfs-gadget gadget: mtp_read(512)
[  267.372000] configfs-gadget gadget: mtp_read: waiting for online state
[  267.632000] dwc2 c0040000.dwc2otg: new device is high-speed
[  267.700000] cfs_uevent_work: sent uevent USB_STATE=CONNECTED
[  267.712000] dwc2 c0040000.dwc2otg: new device is high-speed
[  267.780000] dwc2 c0040000.dwc2otg: new address 38
[  267.804000] configfs-gadget gadget: high-speed config #1: c
[  267.804000] configfs-gadget gadget: mtp_function_set_alt intf: 0 alt: 0
[  267.804000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  267.804000] cfs_uevent_work: sent uevent USB_STATE=CONFIGURED
[  267.864000] configfs-gadget gadget: rx ffffffc023b5b880 16
[  267.864000] configfs-gadget gadget: mtp_read returning 16
[  267.864000] configfs-gadget gadget: mtp_write(12)
[  267.864000] configfs-gadget gadget: mtp_write returning 12
[  267.864000] configfs-gadget gadget: mtp_read(512)
[  267.864000] configfs-gadget gadget: mtp_read: waiting for online state
[  267.864000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  267.864000] configfs-gadget gadget: rx ffffffc023b5b880 12
[  267.864000] configfs-gadget gadget: mtp_read returning 12
[  267.864000] configfs-gadget gadget: mtp_write(349)
[  267.864000] configfs-gadget gadget: mtp_write returning 349
[  267.864000] configfs-gadget gadget: mtp_write(12)
[  267.864000] configfs-gadget gadget: mtp_write returning 12
[  267.868000] configfs-gadget gadget: mtp_read(512)
[  267.868000] configfs-gadget gadget: mtp_read: waiting for online state
[  267.868000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  267.868000] configfs-gadget gadget: rx ffffffc023b5b880 20
[  267.868000] configfs-gadget gadget: mtp_read returning 20
[  267.868000] configfs-gadget gadget: mtp_write(30)
[  267.868000] configfs-gadget gadget: mtp_write returning 30
[  267.868000] configfs-gadget gadget: mtp_write(12)
[  267.868000] configfs-gadget gadget: mtp_write returning 12
[  267.868000] configfs-gadget gadget: mtp_read(512)
[  267.868000] configfs-gadget gadget: mtp_read: waiting for online state
[  267.868000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  267.868000] configfs-gadget gadget: rx ffffffc023b5b880 20
[  267.868000] configfs-gadget gadget: mtp_read returning 20
[  267.868000] configfs-gadget gadget: mtp_write(30)
[  267.868000] configfs-gadget gadget: mtp_write returning 30
[  267.868000] configfs-gadget gadget: mtp_write(12)
[  267.868000] configfs-gadget gadget: mtp_write returning 12
[  267.868000] configfs-gadget gadget: mtp_read(512)
[  267.868000] configfs-gadget gadget: mtp_read: waiting for online state
[  267.868000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  267.868000] configfs-gadget gadget: rx ffffffc023b5b880 20
[  267.868000] configfs-gadget gadget: mtp_read returning 20
[  267.868000] configfs-gadget gadget: mtp_write(30)
[  267.868000] configfs-gadget gadget: mtp_write returning 30
[  267.868000] configfs-gadget gadget: mtp_write(12)
[  267.868000] configfs-gadget gadget: mtp_write returning 12
[  267.868000] configfs-gadget gadget: mtp_read(512)
[  267.868000] configfs-gadget gadget: mtp_read: waiting for online state
[  267.868000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  267.868000] configfs-gadget gadget: rx ffffffc023b5b880 20
[  267.868000] configfs-gadget gadget: mtp_read returning 20
[  267.868000] configfs-gadget gadget: mtp_write(30)
[  267.868000] configfs-gadget gadget: mtp_write returning 30
[  267.868000] configfs-gadget gadget: mtp_write(12)
[  267.868000] configfs-gadget gadget: mtp_write returning 12
[  267.868000] configfs-gadget gadget: mtp_read(512)
[... 323 lines cut ...]
[  274.440000] configfs-gadget gadget: mtp_read(512)
[  274.440000] configfs-gadget gadget: mtp_read: waiting for online state
[  274.440000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  274.444000] configfs-gadget gadget: rx ffffffc023b5b880 12
[  274.444000] configfs-gadget gadget: mtp_read returning 12
[  274.444000] configfs-gadget gadget: mtp_write(20)
[  274.444000] configfs-gadget gadget: mtp_write returning 20
[  274.444000] configfs-gadget gadget: mtp_write(12)
[  274.444000] configfs-gadget gadget: mtp_write returning 12
[  274.444000] configfs-gadget gadget: mtp_read(512)
[  274.444000] configfs-gadget gadget: mtp_read: waiting for online state
[  274.444000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  274.444000] configfs-gadget gadget: rx ffffffc023b5b880 16
[  274.444000] configfs-gadget gadget: mtp_read returning 16
[  274.444000] configfs-gadget gadget: mtp_write(66)
[  274.444000] configfs-gadget gadget: mtp_write returning 66
[  274.444000] configfs-gadget gadget: mtp_write(12)
[  274.444000] configfs-gadget gadget: mtp_write returning 12
[  274.444000] configfs-gadget gadget: mtp_read(512)
[  274.444000] configfs-gadget gadget: mtp_read: waiting for online state
[  274.444000] configfs-gadget gadget: rx ffffffc023b5b880 queue
[  280.124000] dwc2 c0040000.dwc2otg: dwc2_hsotg_ep_stop_xfr: timeout DOEPCTL.EPDisable
[  280.124000] dwc2 c0040000.dwc2otg: GINNakEff triggered
[  280.124000] configfs-gadget gadget: mtp_read returning -512
[  280.132000] mtp_release

gregoiregentil, 2018-11-27:
Same :-( The daemon doesn't crash but nothing happens on Windows.

jfdelnero, 2018-11-29:
I have fixed an issue with the endpoints configuration in FunctionFS mode. The endpoints configuration procedure isn't the same as the gadgetfs mode.
This mistake may cause problems with some usb drivers.
Please update your local repository and let me know if this solve the issue.

jfdelnero, 2018-11-29:
Ok fine ! 
For the other issue can you try to enable the acm gadget without the mtp entry : 
https://github.com/viveris/uMTP-Responder/blob/master/conf/umtprd-ffs.sh
(comment the line 28 &  32 to disable the mtp support).
Tell me if the reconnection/cable insertion is working with the acm gadget.

jfdelnero, 2018-11-30:
I have fixed the FunctionFS mode reconnection issue. Please update from the trunk and try this version.
Now we shouldn't be too far from a new major release with the FunctionFS support. :)

gregoiregentil, 2018-11-30:
Seems working now! I think that it starts to be good. Thanks for your hard fixes. Note that I also managed to make work the ubuntu phablet-team mtp-responder which use the f_mtp.c kernel driver instead of f_ffs.c.

closed 2018-11-30 by gregoiregentil