viveris/uMTP-Responder #14: Adding an option to not show hidden files

gregoiregentil, 2018-12-01:
Here is the patch:

diff --git a/inc/mtp.h b/inc/mtp.h
index 4ffa91f..3b226b7 100644
--- a/inc/mtp.h
+++ b/inc/mtp.h
@@ -67,6 +67,7 @@ typedef struct mtp_usb_cfg_
 
 	int wait_connection;
 	int loop_on_disconnect;
+	int show_hidden_files;
 
 }mtp_usb_cfg;
 
diff --git a/src/fs_handles_db.c b/src/fs_handles_db.c
index 695f353..161e9ba 100644
--- a/src/fs_handles_db.c
+++ b/src/fs_handles_db.c
@@ -376,7 +376,7 @@ int scan_and_add_folder(fs_handles_db * db, char * base, uint32_t parent, uint32
 			PRINT_DEBUG("IsDir: %d",fileinfo.isdirectory);
 			PRINT_DEBUG("---------------------");
 
-			if( strcmp(fileinfo.filename,"..") && strcmp(fileinfo.filename,".") )
+			if( strcmp(fileinfo.filename,"..") && strcmp(fileinfo.filename,".") && (((mtp_ctx *)db->mtp_ctx)->usb_cfg.show_hidden_files || fileinfo.filename[0] != '.') )
 			{
 				add_entry(db, &fileinfo, parent, storage_id);
 			}
diff --git a/src/mtp_cfg.c b/src/mtp_cfg.c
index 4ee460b..604e681 100644
--- a/src/mtp_cfg.c
+++ b/src/mtp_cfg.c
@@ -72,7 +72,8 @@ enum
 	INTERFACE_STRING_CMD,
 
 	WAIT_CONNECTION,
-	LOOP_ON_DISCONNECT
+	LOOP_ON_DISCONNECT,
+	SHOW_HIDDEN_FILES
 };
 
 typedef struct kw_list_
@@ -283,6 +284,9 @@ int get_hex_param(mtp_ctx * context, char * line,int cmd)
 			case LOOP_ON_DISCONNECT:
 				context->usb_cfg.loop_on_disconnect = param_value;
 			break;
+			case SHOW_HIDDEN_FILES:
+				context->usb_cfg.show_hidden_files = param_value;
+			break;
 		}
 	}
 
@@ -360,6 +364,7 @@ kw_list kwlist[] =
 
 	{"wait",                get_hex_param,      WAIT_CONNECTION},
 	{"loop_on_disconnect",  get_hex_param,      LOOP_ON_DISCONNECT},
+	{"show_hidden_files",   get_hex_param,      SHOW_HIDDEN_FILES},
 	{ 0, 0, 0 }
 };
 
@@ -434,6 +439,7 @@ int mtp_load_config_file(mtp_ctx * context)
 
 	context->usb_cfg.wait_connection = 0;
 	context->usb_cfg.loop_on_disconnect = 0;
+	context->usb_cfg.show_hidden_files = 1;
 
 	f = fopen(UMTPR_CONF_FILE,"r");
 	if(f)
@@ -482,6 +488,7 @@ int mtp_load_config_file(mtp_ctx * context)
 
 	PRINT_MSG("Wait for connection : %i",context->usb_cfg.wait_connection);
 	PRINT_MSG("Loop on disconnect : %i",context->usb_cfg.loop_on_disconnect);
+	PRINT_MSG("Show hidden files : %i",context->usb_cfg.show_hidden_files);
 
 	return err;
 }

gregoiregentil, 2018-12-01:
https://pastebin.com/258G3yg5

jfdelnero, 2018-12-01:
Thanks ! The modifications are committed.

closed 2018-12-01 by gregoiregentil