viveris/uMTP-Responder #15: MTP_Responder in Raspbery Pi Zero

prasanthgiri2010, 2018-12-17:
Hi Team
I am able to successfully configure MTP-Responder in RPI Zero W, File Transaction b/w Windows and RPI Zero W happening fine.
if i delete a file in RPI sdcard  from Windows over MTP , file is deleted successfully. But if i delete a file in RPI, still i can see the Deleted file  in Windows. only after stopping and restarting MTP-Responder File deletion is updated in Windows PC. How can i solve this issue ?

jfdelnero, 2018-12-17:
Same issue :
https://github.com/viveris/uMTP-Responder/issues/11

jfdelnero, 2018-12-22:
@prasanthgiri2010 : The event support is now implemented. Please follow up on this issue : 
https://github.com/viveris/uMTP-Responder/issues/11

jfdelnero, 2019-01-10:
umtprd-0.9.7 with MTP events support is now released.

https://github.com/viveris/uMTP-Responder/releases/tag/umtprd-0.9.7

closed 2019-01-10 by jfdelnero