viveris/uMTP-Responder #17: Beaglebone Green Compile Error

TanerKaragol, 2019-02-07:
Hi Team:
I have BBG running kernel 3.8.13 (default kernel comming with the product). I have checkted that FUNCTIONFS and GADGETFS are compiled as a module in kernel.
When I compile uMTP, I got fallowing errors:

cc -o obj/usb_gadget.o src/usb_gadget.c -c -I./inc -lpthread -Wall -O3
src/usb_gadget.c: In function ‘init_usb_mtp_gadget’:
src/usb_gadget.c:706:36: error: ‘FUNCTIONFS_DESCRIPTORS_MAGIC_V2’ undeclared (first use in this function)
src/usb_gadget.c:706:36: note: each undeclared identifier is reported only once for each function it appears in
src/usb_gadget.c:707:36: error: ‘FUNCTIONFS_HAS_FS_DESC’ undeclared (first use in this function)
src/usb_gadget.c:707:36: error: ‘FUNCTIONFS_HAS_HS_DESC’ undeclared (first use in this function)
make: *** [obj/usb_gadget.o] Error 1

Any idea about this errors.
Best regards.

jfdelnero, 2019-02-10:
Yes this kernel doesn't support the new FunctionFS descriptors.
I have fixed this. Update your local repository and compile it with the "OLD_FUNCTIONFS_DESCRIPTORS" option : 

make CFLAGS="-DOLD_FUNCTIONFS_DESCRIPTORS"

TanerKaragol, 2019-02-10:
Compile is OK. But I could not able to run service.
I have used umtprd-ffs.sh in a writeable folder, but it did not work. I have entered each line one by one and 4th line gave me error:

sudo mkdir cfg/usb_gadget/g1
mkdir: cannot create directory `cfg/usb_gadget/g1': No such file or directory

I have used "-p" parameter, but it did not work:
sudo mkdir -p cfg/usb_gadget/g1
mkdir: cannot create directory `cfg/usb_gadget': Operation not permitted

Anyway. I will try up to date BBG firmware image.

P.S: Possible typo error in S98uMTPrd: UMTPRD_DAEMON_PATH=/usr/bin/umtprd[.sh]
There should not be .sh I think.

Regards

TanerKaragol, 2019-02-10:
I have found this link:
https://events.static.linuxfound.org/sites/events/files/slides/USB%20Gadget%20Configfs%20API_0.pdf

Do I need to run at least kernel 3.13?

Regards.

jfdelnero, 2019-02-10:
yes, according to this pdf this appears to be the case.
Have you tried to use the GadgetFS mode ?

TanerKaragol, 2019-02-11:
Hi Team:
I have disabled libcomposite module. I have tryed two ways to disable it:
Since there is no any libcomposite line in am335x_evm.sh, I have closed g_ether and g_multi lines.
 And I have renamed module file libcomposite.ko to libcomposite.ko.org. In two prosedure above lsmod dont show libcomposite.

Then I have added gadgetfs config lines in umtprd.conf and tried to start umtprd. But last outputs of the umtprd:
[uMTPrd - Info] inotify : yes
[uMTPrd - Error] FunctionFS USB Config write error (-1 != 76)
[uMTPrd - Error] init_usb_mtp_gadget init error !
[uMTPrd - Error] USB Init failed !
[uMTPrd - Info] uMTP Responder : Disconnected

It seems trying FunctionFS still.

jfdelnero, 2019-02-11:
change the usb_functionfs_mode config line to :
 
usb_functionfs_mode 0x0

TanerKaragol, 2019-02-13:
Now umtprd does not give error. but just says: [uMTPrd - Info] uMTP Responder : Disconnected
My debian laptop dmesg shows fallowing messages during umtprd run:
[ 8589.048047] usb 3-1: device descriptor read/64, error -110
[ 8592.352120] hub 3-0:1.0: unable to enumerate USB device on port 1
[ 8592.704063] usb 1-3: new high-speed USB device number 74 using ehci_hcd
[ 8593.232050] usb 1-3: device not accepting address 74, error -71
[ 8593.344069] usb 1-3: new high-speed USB device number 75 using ehci_hcd
[ 8594.140090] usb 3-1: new full-speed USB device number 57 using uhci_hcd
[ 8594.196137] hub 3-0:1.0: unable to enumerate USB device on port 1
[ 8594.436051] usb 1-3: new high-speed USB device number 76 using ehci_hcd
[ 8595.592078] usb 3-1: new full-speed USB device number 58 using uhci_hcd
[ 8595.648102] hub 3-0:1.0: unable to enumerate USB device on port 1
[ 8595.888085] usb 1-3: new high-speed USB device number 77 using ehci_hcd
[ 8597.080051] usb 3-1: new full-speed USB device number 59 using uhci_hcd
[ 8597.136089] hub 3-0:1.0: unable to enumerate USB device on port 1
[ 8597.380085] usb 1-3: new high-speed USB device number 78 using ehci_hcd
root@lap:~# 

I have updated libmtp, now latest 1.1.16. mtp-detect command does not show any connected device.

jfdelnero, 2019-02-13:
did you start it with the script ? : 
https://github.com/viveris/uMTP-Responder/raw/master/conf/umtprd_gfs.sh

TanerKaragol, 2019-02-14:
Yes, I have used umtprd_gfs.sh script.
Except I have added "sleep 1" inside while loop. Because 2 days ago When I tried this script, few seconds later CPU activity led of Pi was on always. And Pi did not respond to any thing, event ping. So I thought this loop causing CPU %100 and I putted "sleep 1" inside while loop.

jfdelnero, 2019-02-14:
try to set loop_on_disconnect to 1 in the config file.

jfdelnero, 2019-02-14:
please also note that you need to change the usb_dev_path, usb_epin_path, usb_epout_path and usb_epint_path paths in the config file according to your system in the config file.

TanerKaragol, 2019-02-16:
Nothing changed.
After "mount -t gadgetfs gadgetfs /dev/gadget" this command, "/dev/gadget/musb-hdrc" created automaticly. But other paths do not exist on BBG. I thougt these paths will created after running umtprd, but did not created.
I have copied these lines from BBB user. [#9 Comment](https://github.com/viveris/uMTP-Responder/issues/9#issuecomment-431387570)

usb_dev_path   "/dev/gadget/musb-hdrc"
usb_epin_path  "/dev/gadget/ep1in"
usb_epout_path "/dev/gadget/ep2out"
usb_epint_path "/dev/gadget/ep3in"

-----
Then I deleted "sleep 1" inside loop and run script again. BBG again was stucked. On host computer dmesg showed these lines:

[ 4029.501619] usb 1-3: New USB device found, idVendor=1d6b, idProduct=0100
[ 4029.501629] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4029.501637] usb 1-3: Product: The Viveris Product !
[ 4029.501644] usb 1-3: Manufacturer: Viveris Technologies
[ 4029.501651] usb 1-3: SerialNumber: 01234567

mtp-detect also says "No raw devices found."

jfdelnero, 2019-02-16:
Please rebuild umtprd with the debug output (add CFLAGS="-DUSE_SYSLOG -DDEBUG" to the command line) and post here the full log.

TanerKaragol, 2019-02-16:
sudo ./umtprd_gfs.sh 
FATAL: Module dwc2 not found.
[uMTPrd - Info] uMTP Responder
[uMTPrd - Info] Version: v0.9.7 compiled the Mar  1 2015@20:48:44
[uMTPrd - Info] (c) 2018 - 2019 Viveris Technologies
[uMTPrd - Debug] init_inotify_handler : inotify_fd = 3
[uMTPrd - Info] Add storage root folder - Root Path: /
[uMTPrd - Info] Add storage home folder - Root Path: /home
[uMTPrd - Info] USB Device path : /dev/gadget/musb-hdrc
[uMTPrd - Info] USB In End point path : /dev/gadget/ep1in
[uMTPrd - Info] USB Out End point path : /dev/gadget/ep2out
[uMTPrd - Info] USB Event End point path : /dev/gadget/ep3in
[uMTPrd - Info] USB Max packet size : 0x200 bytes
[uMTPrd - Info] Manufacturer string : Viveris Technologies
[uMTPrd - Info] Product string : The Viveris Product !
[uMTPrd - Info] Serial string : 01234567
[uMTPrd - Info] Interface string : MTP
[uMTPrd - Info] USB Vendor ID : 0x1D6B
[uMTPrd - Info] USB Product ID : 0x0100
[uMTPrd - Info] USB class ID : 0x06
[uMTPrd - Info] USB subclass ID : 0x01
[uMTPrd - Info] USB Protocol ID : 0x01
[uMTPrd - Info] USB Device version : 0x3008
[uMTPrd - Info] USB GadgetFS Mode
[uMTPrd - Info] Wait for connection : 0
[uMTPrd - Info] Loop on disconnect : 0
[uMTPrd - Info] Show hidden files : 1
[uMTPrd - Info] inotify : yes
[uMTPrd - Debug] fill_config_descriptor: (Total Len : 9 + 30 = 39)
[uMTPrd - Debug] 09 02 27 00 01 01 05 80 01 
[uMTPrd - Debug] fill_if_descriptor:
[uMTPrd - Debug] 09 04 00 00 03 06 01 01 06 
[uMTPrd - Debug] fill_ep_descriptor:
[uMTPrd - Debug] 07 05 81 02 00 02 00 
[uMTPrd - Debug] fill_ep_descriptor:
[uMTPrd - Debug] 07 05 02 02 00 02 00 
[uMTPrd - Debug] fill_ep_descriptor:
[uMTPrd - Debug] 07 05 83 03 1C 00 06 
[uMTPrd - Debug] fill_config_descriptor: (Total Len : 9 + 30 = 39)
[uMTPrd - Debug] 09 02 27 00 01 01 04 80 01 
[uMTPrd - Debug] fill_if_descriptor:
[uMTPrd - Debug] 09 04 00 00 03 06 01 01 06 
[uMTPrd - Debug] fill_ep_descriptor:
[uMTPrd - Debug] 07 05 81 02 00 02 00 
[uMTPrd - Debug] fill_ep_descriptor:
[uMTPrd - Debug] 07 05 02 02 00 02 00 
[uMTPrd - Debug] fill_ep_descriptor:
[uMTPrd - Debug] 07 05 83 03 1C 00 06 
[uMTPrd - Debug] fill_dev_descriptor:
[uMTPrd - Debug] 12 01 00 00 06 01 01 00 6B 1D 00 01 08 30 01 02 
[uMTPrd - Debug] 03 01 
[uMTPrd - Debug] init_usb_mtp_gadget :
[uMTPrd - Debug] 00 00 00 00 09 02 27 00 01 01 05 80 01 09 04 00 
[uMTPrd - Debug] 00 03 06 01 01 06 07 05 81 02 00 02 00 07 05 02 
[uMTPrd - Debug] 02 00 02 00 07 05 83 03 1C 00 06 09 02 27 00 01 
[uMTPrd - Debug] 01 04 80 01 09 04 00 00 03 06 01 01 06 07 05 81 
[uMTPrd - Debug] 02 00 02 00 07 05 02 02 00 02 00 07 05 83 03 1C 
[uMTPrd - Debug] 00 06 12 01 00 00 06 01 01 00 6B 1D 00 01 08 30 
[uMTPrd - Debug] 01 02 03 01 
[uMTPrd - Debug] init_usb_mtp_gadget : USB config done
[uMTPrd - Debug] handle_ep0 : Entering...
[uMTPrd - Debug] 1 event(s)
[uMTPrd - Debug] EP0 CONNECT
[uMTPrd - Debug] Select without timeout
[uMTPrd - Debug] 1 event(s)
[uMTPrd - Debug] EP0 DISCONNECT
[uMTPrd - Debug] handle_ep0 : Leaving (ctx->stop=1)...
[uMTPrd - Debug] entering deinit_usb_mtp_gadget
[uMTPrd - Debug] Closing usb device...
[uMTPrd - Debug] leaving deinit_usb_mtp_gadget
[... 163 lines cut ...]
[uMTPrd - Debug] 1 event(s)
[uMTPrd - Debug] EP0 SETUP
[uMTPrd - Debug] Setup request 0x06
[uMTPrd - Debug] Get string id #3 (max length 255)
[uMTPrd - Debug] Found 18 bytes
[uMTPrd - Debug] 12 03 30 00 31 00 32 00 33 00 34 00 35 00 36 00 
[uMTPrd - Debug] 37 00 
[uMTPrd - Debug] Select without timeout
[uMTPrd - Debug] 1 event(s)
[uMTPrd - Debug] EP0 SETUP
[uMTPrd - Debug] Setup request 0x09
[uMTPrd - Debug] Set config value
[uMTPrd - Debug] Init end point /dev/gadget/ep1in (0)
[uMTPrd - Debug] init_ep (0):
[uMTPrd - Debug] 01 00 00 00 07 05 81 02 00 02 00 07 05 81 02 00 
[uMTPrd - Debug] 02 00 
[uMTPrd - Debug] Init end point /dev/gadget/ep2out (1)
[uMTPrd - Debug] init_ep (1):
[uMTPrd - Debug] 01 00 00 00 07 05 02 02 00 02 00 07 05 02 02 00 
[uMTPrd - Debug] 02 00 
[uMTPrd - Debug] Init end point /dev/gadget/ep3in (2)
[uMTPrd - Debug] init_ep (2):
[uMTPrd - Debug] 01 00 00 00 07 05 83 03 1C 00 06 07 05 83 03 1C 

At this point BBG stoped, did not respond to anything.

jfdelnero, 2019-02-23:
Is the BBG completely blocked ?

TanerKaragol, 2019-02-24:
It seams blocked. CPU led is on. I think BBG runing %100 CPU load and it cant respond to any thing.
It does not reply to ping request send by my laptop.  it does not respond to Ctrl+c or any other key compinations.
May be it is better to try new kernel version.

jfdelnero, 2019-07-17:
More success on the new kernel ?

closed 2024-07-08 by jfdelnero