viveris/uMTP-Responder #21: new dir error  on win7(chinese)

Music802, 2019-07-08:
Hi. New dir error on win7(chinese).  is it not support  Chinese?

jfdelnero, 2019-07-11:
We probably need to check/fix the unicode/utf8 support. I will work on this.

Music802, 2019-07-11:
Yes, i have try to fixed  it with iconv. 
can uMTP-Responder support Rename and other featurs like android MTP?

jfdelnero, 2019-07-17:
The new version on the trunk now support properly the unicode files/folders names. 
Please try it and let know if all is working fine.

Music802, 2019-07-18:
Hi，this version works  well  on  win7(chinese files/folders new/copy/delete).    
One more question, Win7 's right-click context menu have not  rename item on umtprd's MTP volume. is  it need supported by umtprd?

jfdelnero, 2019-09-02:
@Music802 : This is now fixed. The rename function is enabled under windows now.

closed 2019-09-02 by jfdelnero

Music802, 2019-09-04:
Cool!, it works well!