viveris/uMTP-Responder #23: MTP Driver load, but no device appear in explorer

SylvainBrisebois, 2019-07-23:
Hi,

I'm trying to run the server on my dev board and I have an issue with the MTP device. The umtpd start correctly, the driver appear in my W10 home device manager correctly (no error in all fields), but no device appear in the file explorer.

I use Buildroot to build umtpd, by using the package : https://stackoverflow.com/questions/54131103/buildroot-github-package-re-extracted-at-every-build

My kernel is : 4.14.73-linux4sam_6.0
My cpu is a :  Atmel Sama5D3

I have enabled gadgetfs in the kernel.

I kept the default config file except for the driver at the end, I comment the linux driver and un-comment the GadgetFS driver for the Sama5D2 (it`s the same for D3 I suppose)

I run manually after the boot up the script : /usr/bin/umtprd_gfs.sh to start the server and it runs without any error. When connected it show : 

gadgetfs: bound to atmel_usba_udc driver
gadgetfs: suspended from state 2
gadgetfs: connected
gadgetfs: configuration #1

Anyone had a similar issue ? Any way to help me debugging this more deeply ?

Thanks a lot,
Sylvain

jfdelnero, 2019-07-23:
Can you rebuild umtprd with the "-DDEBUG" flag and post the debug output ?

SylvainBrisebois, 2019-07-23:
I just re-compile the server with debug option, and here is the syslog output. I hope this can help.

[mtpd-debug log.txt](https://github.com/viveris/uMTP-Responder/files/3423358/mtpd-debug.log.txt)

jfdelnero, 2019-07-23:
Look fine but you don't have any particular event/log when plug/unplug the usb cable ?

SylvainBrisebois, 2019-07-23:
I run mtp-detect under ubuntu16.04 and here is the output : 

libmtp version: 1.1.10
Listing raw device(s)
Device 0 (VID=1d6b and PID=0100) is UNKNOWN in libmtp v1.1.10.
Please report this VID/PID and the device model to the libmtp development team
   Found 1 device(s):
   1d6b:0100 @ bus 2, dev 21
Attempting to connect device(s)
ignoring libusb_claim_interface() = -6PTP_ERROR_IO: failed to open session, trying again after resetting USB interface
LIBMTP libusb: Attempt to reset device
outep: usb_get_endpoint_status(): Device or resource busy
ignoring libusb_claim_interface() = -6LIBMTP PANIC: failed to open session on second attempt
Unable to open raw device 0
OK.

And the dmesg log when I connect the device :

[ 3533.380000] gadgetfs: suspended from state 3
[ 3533.380000] gadgetfs: suspended from state 3
[ 3533.710000] gadgetfs: suspended from state 3
[ 3534.040000] usb3751 0-0068: Charger SDP
[ 3534.040000] gadgetfs: suspended from state 3
[ 3534.040000] power_supply bq24075: External power changed
[ 3534.080000] power_supply bq24075: bq24075_ext_current_limit 500
[ 3534.180000] gadgetfs: disconnected
[ 3534.300000] gadgetfs: bound to atmel_usba_udc driver
[ 3534.310000] gadgetfs: suspended from state 2
[ 3534.580000] gadgetfs: connected
[ 3534.590000] gadgetfs: disconnected
[ 3534.630000] gadgetfs: bound to atmel_usba_udc driver
[ 3534.640000] gadgetfs: suspended from state 2
[ 3534.640000] gadgetfs: suspended from state 2
[ 3535.000000] gadgetfs: connected
[ 3535.010000] gadgetfs: configuration #1

jfdelnero, 2019-07-23:
And at the device side ?

SylvainBrisebois, 2019-07-23:
the dmesg was at device side, the mtp-detect on the linux pc.

here is also the syslog on the device side when I un-plug the re-plug : 

Jan  1 05:42:36 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 05:42:36 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 05:42:36 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 05:42:36 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 05:42:36 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 05:42:36 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 05:42:36 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout
Jan  1 05:42:36 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 05:42:36 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 05:42:36 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout
Jan  1 05:42:36 BrReader user.info kernel: usb3751 0-0068: Charger SDP
Jan  1 05:42:36 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 05:42:36 BrReader user.info kernel: power_supply bq24075: External power changed
Jan  1 05:42:36 BrReader user.info kernel: power_supply bq24075: bq24075_ext_current_limit 500
Jan  1 05:42:37 BrReader user.info kernel: gadgetfs: disconnected
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] EP0 DISCONNECT
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] handle_ep0 : Leaving (ctx->stop=1)...
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] entering deinit_usb_mtp_gadget
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] Closing End Point 0...
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] Closing End Point 1...
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] Closing End Point 2...
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] Closing usb device...
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] leaving deinit_usb_mtp_gadget
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] uMTP Responder : Disconnected
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] inotify_thread : read error -1
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] uMTP Responder
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Version: v0.9.7 compiled the Jul 23 2019@11:49:09
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] (c) 2018 - 2019 Viveris Technologies
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] init_mtp_responder
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] init_inotify_handler : inotify_fd = 4
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] init_mtp_responder : Ok !
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Add storage root - Root Path: /
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] mtp_add_storage : /
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] mtp_add_storage : Storage 00000001 mapped to / (root)
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Add storage media - Root Path: /mediaData
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] mtp_add_storage : /mediaData
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] mtp_add_storage : Storage 00000002 mapped to /mediaData (media)
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB Device path : /dev/gadget/atmel_usba_udc
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB In End point path : /dev/gadget/ep1
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB Out End point path : /dev/gadget/ep2
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB Event End point path : /dev/gadget/ep3
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB Max packet size : 0x200 bytes
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Manufacturer string : Me!
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Product string : MyProduct
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Serial string : 1234
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Interface string : MTP
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB Vendor ID : 0x1D6B
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB Product ID : 0x0100
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB class ID : 0x06
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB subclass ID : 0x01
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB Protocol ID : 0x01
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB Device version : 0x3008
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] USB GadgetFS Mode
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Wait for connection : 0
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Loop on disconnect : 0
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] Show hidden files : 1
Jan  1 05:42:37 BrReader user.notice umtprd: [uMTPrd - Info] inotify : yes
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] fill_config_descriptor: (Total Len : 9 + 30 = 39)
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] 09 02 27 00 01 01 05 80 01
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] fill_if_descriptor:
[... cut ...]
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] EP0 SETUP
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] Setup request 0x06
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] Get string id #6 (max length 255)
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] Found 8 bytes
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] 08 03 4D 00 54 00 50 00
Jan  1 05:42:37 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout

jfdelnero, 2019-07-23:
Try to set loop_on_disconnect to 1.
If still doesn't work you should try the functionfs mode.

SylvainBrisebois, 2019-07-23:
The set loop_on_disconnect to 1 didn't fix the problem.

I tried the functionfs mode, but it seems to not work at all...
I have a lot of errors when I tried to launch the script : /usr/bin/umtprd-ffs.sh

modprobe: module libcomposite not found in modules.dep
file system registered
configfs-gadget 500000.gadget: failed to start g1: -19

Did I have to perform any init before ?
The FunctionFS is loaded in the kernel.

Also, I notice in the script that the libcomposite module is loaded, but it's not in my system. Why it is used for the mtp ? I will try include it in my rootfs right now.

SylvainBrisebois, 2019-07-23:
I found the problem !!!

The usb driver have multiple fifo mode and the default one is not good for mtp (0). For atmel devices we need to set the mode 3 : atmel_usba_udc.fifo_mode=3

Reference : https://www.at91.com/linux4sam/bin/view/Linux4SAM/USBGadgetConfig

So the GadgetFS with mode3 is working perfectly !

Thank you so much for your support !
Very appreciated !

jfdelnero, 2019-07-23:
Perfect ! :)

(yes the libcomposite module is needed for the FunctionFS mode).

closed 2019-07-23 by SylvainBrisebois