viveris/uMTP-Responder #24: Spamming gadgetfs: suspended from state 3

SylvainBrisebois, 2019-07-23:
Hi,

I have an issue with the release 0.9.7.
I have configured the  option with loop_on_disconnect 1 and start umtprd at boot in my init.d

This is working fine, but when I remove the usb device, I got a very fast spamming on the syslog with the message : gadgetfs: suspended from state 3

I tried with UDEV, but in some times this cause problem, since the charging and other functions are on the same port...

I suppose this is due to the re-connection attempt.
Does it's normal ?

How can I change this to avoid flooding my log ?

Thanks for help.
Sylvain

jfdelnero, 2019-07-23:
Have you tried to launch umtprd with the example sh script with loop_on_disconnect set to 0 ?

SylvainBrisebois, 2019-07-24:
Yes, it's the first thing I tried :) It does the same thing. 

Also I notice a weird behaviour, if I set loop_on_disconnect to 0, then plug the usb to my linux pc and then launch umtpd, it connect and then immediatly disconnect and quit. It does this everytime I launch it. When I connect to my w10 pc it require only one time to connect if the usb is already plugged. If i start the bin and then plug the usb, it quit.

If I launch it with the script that reload automatically (umtprd_gfs.sh) , it works, but always require two connection to succeed (see it in the log) : 

 /usr/bin/umtprd;/usr/bin/umtprd
gadgetfs: bound to atmel_usba_udc driver
gadgetfs: suspended from state 2
gadgetfs: connected
gadgetfs: disconnected
gadgetfs: bound to atmel_usba_udc driver
gadgetfs: suspended from state 2
gadgetfs: connected
gadgetfs: configuration #1

I will re-build with the debug print to give you more details, but is seems to be a timing issue or something like that.

and in any case, when I unplug, the span begin : 

 /usr/bin/umtprd;/usr/bin/umtprd
gadgetfs: bound to atmel_usba_udc driver
gadgetfs: suspended from state 2
gadgetfs: connected
gadgetfs: disconnected
gadgetfs: bound to atmel_usba_udc driver
gadgetfs: suspended from state 2
gadgetfs: connected
gadgetfs: configuration #1
gadgetfs: suspended from state 3
bq24075 bq24075: Power removed
bq24075 bq24075: Charge stopped
gadgetfs: suspended from state 3
gadgetfs: suspended from state 3
gadgetfs: suspended from state 3
... for ever until I re-plug or ctrl-c the program

SylvainBrisebois, 2019-07-24:
Here is the debug output I got when I un-plug the device : 

Jan  1 01:54:06 BrReader user.notice root: start here (the moment I unplug the device)
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: bq24075 bq24075: Power removed
Jan  1 01:54:12 BrReader user.info kernel: bq24075 bq24075: Charge stopped
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] Select without timeout
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.debug umtprd: [uMTPrd - Debug] 1 event(s)
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3
Jan  1 01:54:12 BrReader user.info kernel: gadgetfs: suspended from state 3

SylvainBrisebois, 2019-07-24:
I add some debug print and I found by debugging in the code that this case is called when I unplug the device : 
usb_gadget.c -> handle_ep0 -> 

case GADGETFS_SUSPEND:
	PRINT_DEBUG("EP0 SUSPEND");
	break;

From there there is a way to trig a restart of the server or go in an idle mode ?

SylvainBrisebois, 2019-07-24:
I'm working to have the lib libcomposite in my kernel, but it`s seems to be tricky to enable it in my configuration... I wan tto test the ffs mode...

SylvainBrisebois, 2019-07-24:
With the libcomposite in ffs mode, the umtprd work great. So I suppose there is a bug in the gadgetfs driver. I can help troubleshoot if you have any suggestion. For now I will switch to ffs usage since it's working for me.

SylvainBrisebois, 2019-09-10:
Definitively, seems to be a problem in the gadgetfs driver in my system. No fix to be done in the lib. Thanks for your support.

closed 2019-09-10 by SylvainBrisebois