viveris/uMTP-Responder #25: V0.9.7 MTP Chrome-os folder view wrong and file deletion not working

SylvainBrisebois, 2019-07-29:
OS used for test : Chrome-os 74.4.58

The folder listing is very weird, a infinite folder recursive list appear.
The root of the storage is present and I can drop file on it, but I can't delete some files

The same device is running perfectly under Ubuntu18, Windows 10 and MAC 10.14.5

SylvainBrisebois, 2019-07-30:
I just tried with the latest version 0.10.1 and the problem is still the same.

jfdelnero, 2019-07-30:
I think that i will have to get a chrome os device to fix this issue...
Anyway please enable the logs and the result here.

SylvainBrisebois, 2019-07-30:
I use this port : https://www.neverware.com/freedownload/ 
You can run a chrome-os on your regular laptop without installing it.

For the logs, I will get them assap.

SylvainBrisebois, 2019-07-30:
04:55:59 device user.notice root: Starting uMTPrd
04:56:04 device user.notice umtprd: [uMTPrd - Info] uMTP Responder
04:56:04 device user.notice umtprd: [uMTPrd - Info] Version: v0.10.1 compiled the Jul 30 2019@12:32:24
04:56:04 device user.notice umtprd: [uMTPrd - Info] (c) 2018 - 2019 Viveris Technologies
04:56:04 device user.notice umtprd: [uMTPrd - Info] Add storage Media folder - Root Path: /mediaData
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB Device path : /dev/ffs-umtp/ep0
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB In End point path : /dev/ffs-umtp/ep1
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB Out End point path : /dev/ffs-umtp/ep2
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB Event End point path : /dev/ffs-umtp/ep3
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB Max packet size : 0x200 bytes
04:56:04 device user.notice umtprd: [uMTPrd - Info] Manufacturer string : Me
04:56:04 device user.notice umtprd: [uMTPrd - Info] Product string : My precious device
04:56:04 device user.notice umtprd: [uMTPrd - Info] Serial string : 01234567
04:56:04 device user.notice umtprd: [uMTPrd - Info] Interface string : MTP
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB Vendor ID : 0x1D6B
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB Product ID : 0x0100
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB class ID : 0x06
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB subclass ID : 0x01
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB Protocol ID : 0x01
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB Device version : 0x3008
04:56:04 device user.notice umtprd: [uMTPrd - Info] USB FunctionFS Mode
04:56:04 device user.notice umtprd: [uMTPrd - Info] Wait for connection : 0
04:56:04 device user.notice umtprd: [uMTPrd - Info] Loop on disconnect : 1
04:56:04 device user.notice umtprd: [uMTPrd - Info] Show hidden files : 1
04:56:04 device user.notice umtprd: [uMTPrd - Info] inotify : yes
04:56:04 device user.info kernel: read descriptors
04:56:04 device user.info kernel: read strings
04:56:05 device user.info kernel: configfs-gadget gadget: high-speed config #1: c
04:56:06 device user.warn umtprd: [uMTPrd - Warning] MTP_OPERATION_GET_OBJECT_PROP_DESC : Property unsupported ! : 0xDC04 (MTP_PROPERTY_OBJECT_SIZE) (Format : 0x3000 - MTP_FORMAT_UNDEFINED )
04:56:06 device user.warn umtprd: [uMTPrd - Warning] MTP_OPERATION_GET_OBJECT_PROP_DESC : Property unsupported ! : 0xDC04 (MTP_PROPERTY_OBJECT_SIZE) (Format : 0x3001 - MTP_FORMAT_ASSOCIATION )

jfdelnero, 2019-08-01:
I have committed the code to handle these missing properties. Uncomment in mtp_support_def.c MTP_OPERATION_GET_OBJECT_PROP_VALUE.

If not working as expected, comment MTP_OPERATION_GET_OBJECT_PROP_VALUE, MTP_OPERATION_SET_OBJECT_PROP_VALUE, MTP_OPERATION_GET_OBJECT_PROP_DESC and MTP_OPERATION_GET_OBJECT_PROPS_SUPPORTED in  mtp_support_def.c

jfdelnero, 2019-08-23:
I tried it this week and made some fixes. The current trunk version seems to work properly now.

jfdelnero, 2019-09-02:
Closing this issue. Reopen if you still have issue related to this ticket.

closed 2019-09-02 by jfdelnero

SylvainBrisebois, 2019-09-10:
Thanks a lot, the version 0.11.1 fix all the bugs I had !