viveris/uMTP-Responder #35: Add file permissions option

fifteenhex, 2020-04-13:
At the moment files are created with r/w for the owner only. In my use-case I need the files to be readable by other users. Right now I've hacked the code but I think it might be useful to have the file permissions settable from the config file.

jfdelnero, 2020-04-14:
> At the moment files are created with r/w for the owner only. In my use-case I need the files to be readable by other users. Right now I've hacked the code but I think it might be useful to have the file permissions settable from the config file.

Yes we might need this too. I will implement this, probably in the "storage" options.

studiofuga, 2020-06-24:
Thanks, I have a similar use case, but we also need owner changes and per-directory settings.
i.e., server is running as root (changing to other user will not work).
we have 3 directories exported: data, logs and updates.
In data directory, copied files must be owned by user "nobody" with permissions 0600, updates may be changed to another user with permissions 0660. 
We probably work on that in the next weeks and submit a PR eventually. 

Thanks.

studiofuga, 2020-07-07:
See PR #38

closed 2023-03-06 by fifteenhex