viveris/uMTP-Responder #37: File created on PI not showing on Mac Drive

xianrecords2015, 2020-06-11:
I am using the uMTP-Responder on a PI4 that has a USB Flash Drive. The PI is connected to a Mac through the USB-c connection.

I am able to see the drive on both side. I have a mounted drive on the PI and a mounted drive on the MAC. When I create a file on the MAC, I can see it on the drive of the PI.

When I create a file on the PI, it does not show up on the drive of the MAC.

Any idea what is going on ?

Thank you,

Christian

jfdelnero, 2020-06-11:
Interesting. Until now the MTP protocol wasn't supported by Apple at all... Is it something new ? Unless you are speaking about something else ?

xianrecords2015, 2020-06-11:
MTP seems to be supported on MAC. I can see the Drive, I can create files on the MAC drive and see them on the PI. But when I create files on the PI, they are not being seen on the MAC. 
It seems to be the same on Windows.

jfdelnero, 2020-06-11:
which version of macOS are you using ?

uMTP-Responder is using the kernel inotify events to get the file system modifications events. Maybe there is an issue around this ? You will have to enable the debug output and check that uMTP-Responder is "seeing" the PI-side files creation.

xianrecords2015, 2020-06-11:
I am using macOS Catalina 10.15.2. Will check with the debug output

xianrecords2015, 2020-06-11:
BTW, How do you enable the debug output ? Thanks

jfdelnero, 2020-06-11:
Add the CFLAGS="-DDEBUG" option at the make command.

xianrecords2015, 2020-06-11:
I am getting totally confused now. I have removed the umtpd program from being launch at boot in the PI. I am also running a ps -ax to check the processes and do not see it running. I am getting the following:

mkdir: cannot create directory ‘cfg’: File exists
mount: /home/pi/cfg: none already mounted on /home/pi/cfg.
mkdir: cannot create directory ‘cfg/usb_gadget/g1’: File exists
mkdir: cannot create directory ‘configs/c.1’: File exists
mkdir: cannot create directory ‘functions/ffs.mtp’: File exists
mkdir: cannot create directory ‘strings/0x409’: File exists
mkdir: cannot create directory ‘configs/c.1/strings/0x409’: File exists
ln: failed to create symbolic link 'configs/c.1/ffs.mtp': File exists
mkdir: cannot create directory ‘/dev/ffs-mtp’: File exists
mount: /dev/ffs-mtp: mtp already mounted on /dev/ffs-mtp.
[uMTPrd - 10:21:21 - Info] uMTP Responder
[uMTPrd - 10:21:21 - Info] Version: v1.3.6 compiled the Jun 10 2020@20:04:13
[uMTPrd - 10:21:21 - Info] (c) 2018 - 2020 Viveris Technologies
[uMTPrd - 10:21:21 - Info] Add storage usb drive - Root Path: /mnt/usbDrive - Flags: 0x00000000
[uMTPrd - 10:21:21 - Info] USB Device path : /dev/ffs-mtp/ep0
[uMTPrd - 10:21:21 - Info] USB In End point path : /dev/ffs-mtp/ep1
[uMTPrd - 10:21:21 - Info] USB Out End point path : /dev/ffs-mtp/ep2
[uMTPrd - 10:21:21 - Info] USB Event End point path : /dev/ffs-mtp/ep3
[uMTPrd - 10:21:21 - Info] USB Max packet size : 0x200 bytes
[uMTPrd - 10:21:21 - Info] USB Max write buffer size : 0x10000 bytes
[uMTPrd - 10:21:21 - Info] USB Max read buffer size : 0x10000 bytes
[uMTPrd - 10:21:21 - Info] Read file buffer size : 0x100000 bytes
[uMTPrd - 10:21:21 - Info] Manufacturer string : Viveris Technologies
[uMTPrd - 10:21:21 - Info] Product string : Viveris Technologies
[uMTPrd - 10:21:21 - Info] Serial string : 01234567
[uMTPrd - 10:21:21 - Info] Interface string : MTP
[uMTPrd - 10:21:21 - Info] USB Vendor ID : 0x1D6B
[uMTPrd - 10:21:21 - Info] USB Product ID : 0x0100
[uMTPrd - 10:21:21 - Info] USB class ID : 0x06
[uMTPrd - 10:21:21 - Info] USB subclass ID : 0x01
[uMTPrd - 10:21:21 - Info] USB Protocol ID : 0x01
[uMTPrd - 10:21:21 - Info] USB Device version : 0x3008
[uMTPrd - 10:21:21 - Info] USB FunctionFS Mode
[uMTPrd - 10:21:21 - Info] Wait for connection : 0
[uMTPrd - 10:21:21 - Info] Loop on disconnect : 0
[uMTPrd - 10:21:21 - Info] Show hidden files : 1
[uMTPrd - 10:21:21 - Info] inotify : yes
ls: write error: Device or resource busy

And for some reason, the Drive is mounted on my MAC and when I add file on my MAC, I get notification on the PI:

[uMTPrd - 10:22:01 - Info] uMTP Responder : Disconnected

Not understanding what is going on

xianrecords2015, 2020-06-11:
I run it with gadgetfs on the PI and this is what I get:

[uMTPrd - 10:38:50 - Info] uMTP Responder
[uMTPrd - 10:38:50 - Info] Version: v1.3.6 compiled the Jun 10 2020@20:04:13
[uMTPrd - 10:38:50 - Info] (c) 2018 - 2020 Viveris Technologies
[uMTPrd - 10:38:50 - Info] Add storage usb drive - Root Path: /mnt/usbDrive - Flags: 0x00000000
[uMTPrd - 10:38:50 - Info] USB Device path : /dev/ffs-mtp/ep0
[uMTPrd - 10:38:50 - Info] USB In End point path : /dev/ffs-mtp/ep1
[uMTPrd - 10:38:50 - Info] USB Out End point path : /dev/ffs-mtp/ep2
[uMTPrd - 10:38:50 - Info] USB Event End point path : /dev/ffs-mtp/ep3
[uMTPrd - 10:38:50 - Info] USB Max packet size : 0x200 bytes
[uMTPrd - 10:38:50 - Info] USB Max write buffer size : 0x10000 bytes
[uMTPrd - 10:38:50 - Info] USB Max read buffer size : 0x10000 bytes
[uMTPrd - 10:38:50 - Info] Read file buffer size : 0x100000 bytes
[uMTPrd - 10:38:50 - Info] Manufacturer string : Viveris Technologies
[uMTPrd - 10:38:50 - Info] Product string : Viveris Technologies
[uMTPrd - 10:38:50 - Info] Serial string : 01234567
[uMTPrd - 10:38:50 - Info] Interface string : MTP
[uMTPrd - 10:38:50 - Info] USB Vendor ID : 0x1D6B
[uMTPrd - 10:38:50 - Info] USB Product ID : 0x0100
[uMTPrd - 10:38:50 - Info] USB class ID : 0x06
[uMTPrd - 10:38:50 - Info] USB subclass ID : 0x01
[uMTPrd - 10:38:50 - Info] USB Protocol ID : 0x01
[uMTPrd - 10:38:50 - Info] USB Device version : 0x3008
[uMTPrd - 10:38:50 - Info] USB FunctionFS Mode
[uMTPrd - 10:38:50 - Info] Wait for connection : 0
[uMTPrd - 10:38:50 - Info] Loop on disconnect : 0
[uMTPrd - 10:38:50 - Info] Show hidden files : 1
[uMTPrd - 10:38:50 - Info] inotify : yes
[uMTPrd - 10:38:50 - Error] init_usb_mtp_gadget : Unable to open /dev/ffs-mtp/ep0 (No such file or directory)
[uMTPrd - 10:38:50 - Error] init_usb_mtp_gadget init error !
[uMTPrd - 10:38:50 - Error] USB Init failed !
[uMTPrd - 10:38:50 - Info] uMTP Responder : Disconnected
[uMTPrd - 10:38:50 - Error] Error : Couldn't run the main thread... (-2)

It is looping through that.

jfdelnero, 2020-06-11:
>[uMTPrd - 10:38:50 - Info] USB FunctionFS Mode

uMTPrd is FunctionFS mode. check your config file.

EricPreventa, 2020-11-30:
Not sure how you were able to get the Pi4 recognized on a Mac. I was not able to. Are you sure you aren't "by accidentically" connecting it via mass storage or samba?

jfdelnero, 2020-11-30:
Yes i agree that this issue is not very clear. I know that there is some specific softwares able to connect to MTP device, but until now MTP devices wasn't supported at the operation system/macOS level.
Maybe the last OS versions now support the MTP ? I can't check since my machine is limited to High Sierra.