viveris/uMTP-Responder #43: 5.x.x kernels not supported?

jeffbronte, 2020-10-07:
I know release notes have 4.x.x testing, just asking.
Also, in mtp_incoming_packet(), if the buffer doesn't match header, the buffer size is used rather than the header size, (I know this because the 5.4 kernel packets are coming out mismatched. This induces errors. 
IMO  
     size = mtp_packet_hdr->length;
when this condition occurs...

wish I had time to help, I'll ask the boss...

jfdelnero, 2020-10-08:
5.x.x should be supported. If you have any issue with this kernel version let me know !

(btw i will probably update kernel of my test platform to a 5.4 version).

jeffbronte, 2020-10-08:
I'm on a Xilinx Petalinux Ultrascale+ system, kernel is 5.4.0, f_fs.c is unchanged from kernel.org, but maybe something under the covers? I have a version of umtprd with debug and can send you the traces. I was able to get mine to work by taking my 4.19 kernel fs.c, (same target), and migrating it forward (compiles no changes, thankfully). It does seem to work perfectly using the 4.19 usb_f_fs.ko compiled for 5.4. I'll post something later today.

jeffbronte, 2020-10-12:
Attached are logs with umtprd compiled with debug. There is a minor change, where in mtp_incoming_packet, the header size is used rather than the actual size, otherwise these logs explode (printing the 65K invalid packet from 5.4 f_fs.c). Again, the f_fs.c from 4.19 will compile and work with the latest umtprd, when dropped into 5.4 kernel build. 

[kernel-5.4.log](https://github.com/viveris/uMTP-Responder/files/5366028/kernel-5.4.log)

[kernel-4.19.log](https://github.com/viveris/uMTP-Responder/files/5366029/kernel-4.19.log)

jfdelnero, 2020-10-12:
Can you try to uncomment the usb_max_rd_buffer_size, usb_max_wr_buffer_size and  read_buffer_cache_size lines in the config file ?

jeffbronte, 2020-10-12:
Sure, when I have a chance to revert my working target back to broken

jfdelnero, 2020-10-12:
This page is quite interesting since this seems to be the same buffer size issue :  

https://www.toradex.com/community/questions/39123/usb-configfs-functionfs-read-of-endpoint-bundles-p.html

> In the meantime I discovered the following thing: The size of my receive buffer is 64 kB aligned to 4 bytes.
> 
> When I reduce this size to 20 kB then the read() returns after every sent packet as it should be.
> 
> When I change the buffer size to 21 kB then the read () returns after every second sent packet.
> 
> Is there a limitation in the linux usb driver stack I have to consider?

And apparently the Xilinx driver appears to be limited to 7KB (see USB_BUFSIZE, not sure if this the same value on your platform)

https://forums.xilinx.com/xlnx/attachments/xlnx/ELINUX/29418/1/usb.c

The default max buffer size is currently 64KB in uMTP Responder. This definitely appears to be too much for some plateforms (iMX and Xilinx platforms maybe). I will reduce this default value or find a way to get the max USB authorized buffer size on the target.

#36 

#27

jeffbronte, 2020-10-12:
Ok, but I think could be more to it. When I used the message size instead of the buffer size there are more errors, by diffing 4.19 and 5.4 you can see the changes in standard driver file ops. Here's a patch that does work. 4.19 -> 5.4
[0005-FunctionFS.patch.zip](https://github.com/viveris/uMTP-Responder/files/5367206/0005-FunctionFS.patch.zip)

jeffbronte, 2020-12-28:
> Can you try to uncomment the usb_max_rd_buffer_size, usb_max_wr_buffer_size and read_buffer_cache_size lines in the config file ?

Follow up: uncommenting these buffer sizes to the defaults works, 5.4.0 on Xilinx Ultrascale+ (Petalinux) works. Issue is closed.

closed 2020-12-28 by jeffbronte

jfdelnero, 2020-12-28:
Note : I have changed/reduced the default built-in values some weeks ago. I think that this should work without changing the configuration with the current master repository.