viveris/uMTP-Responder #45: Documentation on Compiling for Raspberry pi

EricPreventa, 2020-11-26:
I didn't see any documentation as to how to compile this for a RP. New at this stuff, my apologies. Would love to try it.

jfdelnero, 2020-11-26:
the "make" command should be enough to build it.

EricPreventa, 2020-11-29:
My apologies @jfdelnero . I've clarified the title better.

I'm looking to use/run this on a raspberry pi 4b.

1. I have successfully built this with debug mode using the make command
2. I have copied the file in conf/umtprd.conf to /etc/umtprd/umtprd.conf
3. I have uncommented the lines in umtprd.conf where it says "GadgetFS Mode : Raspberry PI Zero/..."
4. When I try running in conf the following files (umtprd-ffs.sh, umtprd_gfs.sh), I either get an error saying it's been mounted or it just sits there and hangs (I've waited a few minutes and nothing happened)

At this point my questions are:
A. What have I done wrong above to not succeed in running this
B. What do I need to do
C. What is the difference between umtprd-ffs.sh and umtprd_gfs? Is one better than the other for a PI?

Thanks :)

jfdelnero, 2020-11-29:
umtprd-ffs.sh is for the FunctionFS mode. Since you choose to run it in GadgetFS, don't run this script.

umtprd_gfs.sh : If the all the needed stuff started successfully and the GadgetFS driver is running then uMTP is just waiting for a connection. Have you connected the device port to a PC ?
Can you share the logs ?

> C. What is the difference between umtprd-ffs.sh and umtprd_gfs? Is one better than the other for a PI?

umtprd-ffs.sh : FunctionFS mode : The device usb port can be shared with others functions. Example : uMTP and a virtual serial port or any other USB device type.
 
umtprd-gfs.sh : GadgetFS mode - only one USB function at a time (here uMTP). Probably easier to setup for a first try.

EricPreventa, 2020-11-30:
Thank you @jfdelnero I should clarify that for whatever reason...I haven't been able to get either one to work.

I'm connecting my PI to usb-c for both connectivity and power. So it's connected and delivering power. Using mass storage I don't get any problems connecting. 

`sudo ./umtprd_gfs.sh`
`mkdir: cannot create directory ‘/dev/gadget’: File exists
mount: /dev/gadget: gadgetfs already mounted on /dev/gadget.`
this just hangs as it's looking to do something.

If I unmount /dev/gadget and do the following, I get:

`sudo ./umtprd_gfs.sh `
`mkdir: cannot create directory ‘/dev/gadget’: File exists
User defined signal 1`

Once again this just hangs as well. 

Where am I messing up?

EricPreventa, 2020-11-30:
Just built again without debug and got the following error message when running sudo ./umtprd_gfs.sh

[uMTPrd - 06:04:08 - Info] uMTP Responder
[uMTPrd - 06:04:08 - Info] Version: v1.3.10 compiled the Nov 30 2020@06:02:54
[uMTPrd - 06:04:08 - Info] (c) 2018 - 2020 Viveris Technologies
[uMTPrd - 06:04:08 - Info] Add storage root folder - Root Path: / - Flags: 0x00000000
[uMTPrd - 06:04:08 - Info] Add storage home folder - Root Path: /home - Flags: 0x00000001
[uMTPrd - 06:04:08 - Info] Add storage www folder - Root Path: /www - Flags: 0x00000003
[uMTPrd - 06:04:08 - Info] USB Device path : /dev/gadget/20980000.usb
[uMTPrd - 06:04:08 - Info] USB In End point path : /dev/gadget/ep1in
[uMTPrd - 06:04:08 - Info] USB Out End point path : /dev/gadget/ep2out
[uMTPrd - 06:04:08 - Info] USB Event End point path : /dev/gadget/ep3in
[uMTPrd - 06:04:08 - Info] USB Max packet size : 0x200 bytes
[uMTPrd - 06:04:08 - Info] USB Max write buffer size : 0x2000 bytes
[uMTPrd - 06:04:08 - Info] USB Max read buffer size : 0x2000 bytes
[uMTPrd - 06:04:08 - Info] Read file buffer size : 0x100000 bytes
[uMTPrd - 06:04:08 - Info] Manufacturer string : Viveris Technologies
[uMTPrd - 06:04:08 - Info] Product string : Viveris Technologies
[uMTPrd - 06:04:08 - Info] Serial string : 01234567
[uMTPrd - 06:04:08 - Info] Firmware Version string : Rev A
[uMTPrd - 06:04:08 - Info] Interface string : MTP
[uMTPrd - 06:04:08 - Info] USB Vendor ID : 0x1D6B
[uMTPrd - 06:04:08 - Info] USB Product ID : 0x0100
[uMTPrd - 06:04:08 - Info] USB class ID : 0x06
[uMTPrd - 06:04:08 - Info] USB subclass ID : 0x01
[uMTPrd - 06:04:08 - Info] USB Protocol ID : 0x01
[uMTPrd - 06:04:08 - Info] USB Device version : 0x3008
[uMTPrd - 06:04:08 - Info] USB GadgetFS Mode
[uMTPrd - 06:04:08 - Info] Wait for connection : 0
[uMTPrd - 06:04:08 - Info] Loop on disconnect : 0
[uMTPrd - 06:04:08 - Info] Show hidden files : 1
[uMTPrd - 06:04:08 - Info] File creation umask : System default umask
[uMTPrd - 06:04:08 - Info] inotify : yes
[uMTPrd - 06:04:08 - Error] init_usb_mtp_gadget : Unable to open /dev/gadget/20980000.usb (No such file or directory)
[uMTPrd - 06:04:08 - Error] init_usb_mtp_gadget init error !
[uMTPrd - 06:04:08 - Error] USB Init failed !
[uMTPrd - 06:04:08 - Info] uMTP Responder : Disconnected
[uMTPrd - 06:04:08 - Error] Error : Couldn't run the main thread... (-2)

EricPreventa, 2020-11-30:
I was able to solve this problem. I'm using the Rasberry Pi 4 which appears to not have the 20980000.usb instead I changed it in the config file to the fe980000.usb. In addition, I'm using a Mac which apparently doesn't have support for it, however using Windows it recognized it.

closed 2020-11-30 by EricPreventa