viveris/uMTP-Responder #47: Add connection authorization?

joshbowyer, 2020-12-05:
Currently umtp-responder is very permissive and allows connections without any authorization (if I am mistaken please correct me). Can authorization functionality be added similarly to how Android handles it, by requiring some sort of "Allow This Device To Connect" dialogue on the host?

jfdelnero, 2020-12-08:
Yes this is something possible. You can already use the message queue to mount/umount mtp drives from the device. 

I may also add a specialized message queue command to disconnect/reconnect the mtp connection like Android does.

Anyway the autorization user interface on the device is out of the scope of the uMTP responder. It will just provide the message queue interface to unlock/lock the mtp drives.

jfdelnero, 2020-12-08:
Duplicate of #32

jfdelnero, 2021-12-05:
The unlock/lock feature was implemented in the v1.5.1.

joshbowyer, 2021-12-06:
Any documentation on how to enable it and what the scope of it is?

jfdelnero, 2021-12-06:
The readme and the configuration file was updated today. (The official release was done 1 hour ago).