viveris/uMTP-Responder #48: Add option to force uid/gid

joshbowyer, 2020-12-13:
There should be a configurable option in umtprd.conf to force uid/gid for files written to mount loops, such as samba does with the `force user` and `force group` directives. As in, if I specify `force user = 1000` for the /home loop, then any files written via to /home via umtp should have 1000:1000 ownership (which would translate to a user/group on the host system)

jfdelnero, 2020-12-21:
We will consider this feature and study how this can be implemented.

jfdelnero, 2021-12-08:
@joshbowyer  The UID / GID can now be forced for each storage unit. 
currently on the master branch. Configuration file updated showing the "default_uid/default_gid" and "uid=/gid=" usage.

closed 2026-11-01 by jfdelnero