viveris/uMTP-Responder #51: modified file on the mtp responder side, but the PC can not refresh at the same time.

woshizhy, 2020-12-25:
for example:
1）create a file on mtp responder：
    echo hello1 > hello.txt

2）watch the file on PC，it's good to find the hello.txt and it's content is : hello1.

3）append some content to the hello.txt file on mtp responder：
    echo hello2 >> hello.txt

4）watch the file on PC，but the content  is keep to : hello1. 

5）append some content to the hello.txt file on mtp responder：
    echo hello3 >> hello.txt

6）watch the file on PC，and the content  is change to : hello. hello2.

....

can you verification this problem please?

Thank you, good life to you.

jfdelnero, 2020-12-27:
Hello,

Thanks for the patch, this appears fine.

Have you successfully tried it ? If so, please make a pull-request. 

Thanks !

woshizhy, 2020-12-28:
Hello.

First, thank you for your reply.

The  client operating system I have : win7, win10, and my android-10 phone.

And I passed the verification on win7 and win10, but the android-10 phone is not successfully.

woshizhy, 2020-12-28:
Hello.

And I tried the following experiments：
using another android phone as the MTP responder to connect my android-10 phone,
and whether it's creating a new file or modifying a file on the MTP responder, my android-10 do not update.
(Note: if using your uMTP-Responder for the experiment, when I  created a new file or modify a file at mtp responder side, it always take the initiative to report related events to the MTP Initiator. And I know that situation by open the debug log.)

So, I believe that the android os system support for MTP Initiator is not perfect, and it's not about your MTP responder.

Do you agree?

Thanks, wish you a happy life.

jfdelnero, 2020-12-28:
Yes i agree. I know that some MTP Initiators have some issues with the events coming from the device.

Thanks for the PR/patch and tests, this is now in the sources !

closed 2020-12-28 by jfdelnero