viveris/uMTP-Responder #53: The MTP responder side renames the file, but the MTP Initiator  side does not update it

woshizhy, 2020-12-29:
Hello.

As shown in the title, I found that MTP responder did not listen for rename events. (for example: mv hello.c hello.java)
(Only the events of adding/modifying  and deleting files are monitored for your uMTP-Responder)

So when renaming a file on the MTP responder side, no event occurred, and no event was reported to MTP Initiator .

If want to support this function( renames the file on the MTP responder side, and the MTP Initiator also can update)
I find it possible to do this:
1) Add events to listen for renamed files: (modify file: src/inotify.c )
    
```
int inotify_handler_addwatch( mtp_ctx * ctx, char * path )
 {
...
            if ( !ctx->no_inotify )
            {
--                   return inotify_add_watch( ctx->inotify_fd, path, IN_CREATE | IN_DELETE | IN_MODIFY );
++                 return inotify_add_watch( ctx->inotify_fd, path, IN_CREATE | IN_DELETE | IN_MODIFY | 
++                                                                                            IN_MOVED_FROM |  IN_MOVED_TO );
            }
...
 }
```

2) Handling events: (modify file: src/inotify.c )
```
void *inotify_gotsig(int sig, siginfo_t *info, void *ucontext)
{
...
--              if ( event->mask & IN_CREATE )
++            if ( event->mask & IN_CREATE || event->mask & IN_MOVED_TO )
      
--              if ( event->mask & IN_DELETE )
++            if ( event->mask & IN_DELETE || event->mask & IN_MOVED_FROM )
...
 }
```

And I I tested it and it worked. 
(The MTP Initiator tested are: win7 and win 10)

But I just passed the preliminary verification, and I don't know if there are any bugs.

Do you argee?

Thanks, I wish you a happy life.

jfdelnero, 2020-12-29:
Yes this appears fine to me.
Please make a pull request for this one too.
Thanks !

jfdelnero, 2021-01-03:
I have just tested the modification.
Please note that there was a mistake in the patch (IN_CREATE used instead of IN_DELETE in the delete case).
This is now fixed and all appears to work fine.
Thanks !

closed 2021-01-03 by jfdelnero

woshizhy, 2021-01-04:
Oh, I'm sorry I didn't handle it.

Thank you for wiping my ass. ☺

huangjie18, 2021-01-04:
Oh, It's very useful to me. Thank you for solving the problem. @woshizhy