viveris/uMTP-Responder #55: SuperSpeedPlus support?

FrozenSource, 2021-02-08:
Hello,
First let me tell you that I love your work! I have a question however? Is there any ambition as to be compatible with the SuperSpeedPlus protocol? I wish to work with the responder on a pi4 (USB-C). With as much speed as possible...
Sincerely,
Bram Nijenkamp

jfdelnero, 2021-02-08:
SuperSpeed is already supported :  uncomment the "#define CONFIG_USB_SS_SUPPORT 1" line in buildconf.h
(tested in FunctionFs mode with dwc3).

For the SuperSpeedPlus i have read that this is just a flag or something similar to pass to usb the descriptor so let me know if you are interested to make some tests.

FrozenSource, 2021-02-08:
Ok will do that! thanks for telling me. I do need to do some performance testing for a project, so will specify the results when I get to testing!

FrozenSource, 2021-02-23:
I just got to this problem but the Pi 4 apparently has USB 2.0 data lines so I would first need to find a solution to this.
I do have another question though. Does umtpr support password-controlled connections?

closed 2026-11-01 by jfdelnero