viveris/uMTP-Responder #56: Problem with multipe storages a pointed to folder on the same partition.

KonstantinBYvitebsk, 2021-02-08:
Hi. 
I use last software at master branch. Revision. b38b8fb43  Device imx6 (Linux 4.14.98).
When I use two or three storage for example: 

```
storage "/media/extern/update/"      "update" "rw"
storage "/media/extern/media/video/"      "video" "rw"
storage "/media/extern/media/photo"      "photo" "rw"

```

It is the same partition. But different folders:
```
cat /proc/mounts |grep extern
/dev/mmcblk0p6 **/media/extern** vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro 0 0
```
My host is ubuntu 18. 
How to reproduce:
1. Copy big file (90Mb) on device. Sometimes I see the problem on this step. 
2. Copy the file again to the same folder (replace): 
the file is coppied but I see the windows with text which means: "There is a copy error" and "Connection lost". 

There is no errors on umrtpd side.  But on the PC side:
[ 7472.130669] pool[22616]: segfault at 0 ip 00007ff7a2cdca37 sp 00007ff79e740b70 error 4 in libmtp.so.9.3.0[7ff7a2ca5000+5e000]
[ 7472.130714] Code: 85 11 03 00 00 48 81 c4 88 00 00 00 5b 5d 41 5c 41 5d 41 5e 41 5f c3 0f 1f 40 00 41 80 7c 24 04 0f 48 8b 5c 24 48 8b 6c 24 40 <8b> 03 74 02 0f c8 89 44 24 30 48 8b 44 24 08 45 31 db c7 44 24 44
[image]
The problem now only with the "update" storage. Whe I try to copy the file to "photo" storage the problem is not appearing.

Another strange point with WinDev2021Eval on VirtualBox. I see high cpu and mem load when connecting device. May be it is not related.

KonstantinBYvitebsk, 2021-02-08:
> Another strange point with WinDev2021Eval on VirtualBox. I see high cpu and mem load when connecting device. May be it is not related.

My fellow have not reproduced the bug. May be something wrong with my VirtualBox machine.

jfdelnero, 2021-02-09:
This appears to be a libmtp client/mtp initiator crash/issue.
 
> My fellow have not reproduced the bug on real PC. May be something wrong with my VirtualBox machine.

You mean the whole client/mtp initiator issue or the just the VirtualBox machine issue ?

KonstantinBYvitebsk, 2021-02-10:
> You mean the whole client/mtp initiator issue or the just the VirtualBox machine issue ?

It looks like a VirtualBox machine issue.

KonstantinBYvitebsk, 2021-02-10:
I will try to reproduce the bug on different system. Can you keep this issue as opened for a while? I will add here new information if I find. I think it willl take at leas a month.

KonstantinBYvitebsk, 2021-03-12:
It looks like spontaneous error on Ubuntu side. The issue can be closed.

closed 2021-03-12 by KonstantinBYvitebsk

jfdelnero, 2021-04-01:
@KonstantinBYvitebsk  This may have a connection with your issue :
https://github.com/viveris/uMTP-Responder/pull/62

KonstantinBYvitebsk, 2021-04-01:
@jfdelnero It looks like you are right. But now I can not to test it, unfortunately.