viveris/uMTP-Responder #60: Thumbnails

pcercuei, 2021-03-31:
It should be possible to support thumbnails by implementing the "Representative Sample" object properties.

The thumbnailing process itself should be deferred to an external program, pointed to in the config.

jfdelnero, 2021-07-02:
@pcercuei which way should we use to pass the thumbnail image data to the uMTP daemon ?

pcercuei, 2021-07-09:
@jfdelnero there is a Freedesktop standard for that; The directory `/usr/share/thumbnailers` contains *.thumbnailer files with the following schema (example):
```
[Thumbnailer Entry]
TryExec=/usr/bin/gdk-pixbuf-thumbnailer
Exec=/usr/bin/gdk-pixbuf-thumbnailer -s %s %u %o
MimeType=image/svg+xml;image/svg+xml-compressed;
```
The Exec= line tells you how the thumbnailer should be called, (%s: size; %u: url/filepath; %o: output file). It will write the thumbnail to the given output PNG file.

However I believe it's not umtprd's job to create the thumbnails (there is probably an existing daemon for that). Umtprd should only have to open the thumbnails from `~/.cache/thumbnails/normal/{md5 of file}.png`, and if it exist, use this PNG as the thumbnail for the MTP object.