viveris/uMTP-Responder #65: creating and deleting ".." folder wipes storage

skandigraun, 2021-09-12:
Creating a folder with the name of ".." (without quotes), and then deleting it will delete the shared storage.

Reproduction:
1. Connect the device to a Windows PC, and open the shared folder (could work in Linux/Mac also, not sure. Haven't tried.). Have some random files/folders in the storage.
2. On the Windows machine, in the root folder of the MTP device create a directory called ".." (without quotes). This will create a folder called "New Folder".
3. Delete the newly created folder in Windows.

Result: the whole content of storage is wiped, including the random files you created in step 1.

jfdelnero, 2021-09-12:
Thanks for reporting/pointing this security issue.
Your scenario case is now fixed.

skandigraun, 2021-09-12:
Thank you! Will be able to test it and close tomorrow.

closed 2021-09-13 by skandigraun