viveris/uMTP-Responder #73: uMTP on Raspberry Pi 4 Freezing/Crashing Issues While Running Camera Scripts

AndrewLloydSmith, 2021-12-14:
I am working with uMTP on a Raspberry Pi 4 model B, and having some issues with freezing/failing in early operation. I am attempting to access files from a uMTP connected Windows machine while running a image capture/camera script on the Pi. I have found that unless I go very slowly with the first couple of sequences (boot-capture image-access image via uMTP-capture next image-access next image via uMTP...) uMTP will freeze or fail entirely on the second file access. Once I've gotten past the first few captures, everything works as expected and I can capture/copy images as quickly as they are produced. I have had more severe issues when working with other image capture libraries (raspiraw causes uMTP to fail regularly, but I am able to get things generally functioning in my current setup using libcamera-vid).

I am not sure what is causing this, and would greatly appreciate any insight. To give you some idea of the exact setup I'm using, here is a complete list of steps I took to arrange everything: (note that I am using Linux 11 (Bullseye))

#Create sd card w/ Imager 1.6.2

#Select 'update' during install, compelte install/update

reboot

#uMTP-responder

sudo git clone https://github.com/viveris/uMTP-Responder
cd uMTP-Responder
sudo make
cd conf

sudo nano umtprd.conf
#comment out all 'Function FS' lines
#uncomment all 'Gadget FS' lines
#modify 20980000.usb to fe980000.usb

sudo cp umtprd.conf /etc/umtprd/umtprd.conf
sudo cp /home/pi/uMTP-Responder/umtprd /usr/bin/umtprd

sudo nano /boot/config.txt
#append 'dtoverlay=dwc2'

sudo nano /etc/modules
#append 'dwc2'

reboot

#uMTP all setup, launch by running  /conf/umtprd.init

#Run uMTP at boot using rc.local

sudo nano /etc/rc.local
#add 'sudo ./home/pi/uMTP-Responder/conf/umtprd.init start &' right before 'exit 0'

#modify rc.local to call script which performs touch/rm files, triggered by GPIO. Works perfectly.

#Now to get the camera setup alongside

sudo raspi-config >> enable camera

reboot

#modify rc.local to call the following script:
::::::::::::::::::::::::::::::::::::::::::

sudo nice -n -20 /home/pi/uMTP-Responder/conf/umtprd.init start
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/pi:/home/pi/capture_imx:/home/pi/raspiraw:/home/pi/raspiraw/tools

echo "22" > /sys/class/gpio/export
echo "in" > /sys/class/gpio/gpio22/direction
while true
do
sleep 0.1
controlstate=$(cat /sys/class/gpio/gpio22/value)
if [ "$controlstate" -eq 1 ] || [ -f /dev/shm/trigger.txt ]; then

libcamera-vid -t 1000 --docec mjpeg --segment 1 -o /dev/shm/test$05d.jpeg

fi

done

:::::::::::::::::::::::::::::::::::::::

#Now uMTP has a significant change of lock/failure on the first couple captures taken and accessed. If it makes it through the first ~3 runs, it is stable.

jfdelnero, 2021-12-14:
Hello,

Have you the same issue with the last tagged/released version ?

AndrewLloydSmith, 2021-12-15:
Yes, I believe so. In order to make sure I was using the latest code I deleted /home/pi/uMTP-Responder and repeated the git clone, make, and configuration steps. After doing so I still observed the behavior when using libcamera-vid.

I will repeat this test with the other image capture libraries to make sure I didn't miss something, and confirm the result shortly.

jfdelnero, 2021-12-15:
please try these tags instead :

https://github.com/viveris/uMTP-Responder/releases/tag/umtprd-1.5.1

https://github.com/viveris/uMTP-Responder/releases/tag/umtprd-1.3.10

AndrewLloydSmith, 2021-12-16:
I can confirm that this behavior is observable when using the source tar.gz files for both 1.5.1 and 1.3.10

jfdelnero, 2021-12-16:
Can you run valgrind with the latest version available on the current master branch ?

jfdelnero, 2021-12-17:
And also try to disable the inotify feature in the configuration file.

AndrewLloydSmith, 2021-12-29:
The behavior I'm seeing is definitely an effect of inotify. With inotify disabled the system runs as expected, with the caveat that file changes on the Pi 4 may not be recognized by the host computer.

jfdelnero, 2021-12-29:
Thanks for the test !

Can you rebuild umtprd with this flag and post back the logs right before the freeze here ? :
  
make CFLAGS="-DUSE_SYSLOG -DDEBUG"

AndrewLloydSmith, 2021-12-30:
I realize I did not describe this well initially, but the host computer experiences the service freeze/crash. The uMTP process running on the Raspberry Pi does not crash, though it does take up far more of the CPU capacity (~6%) when in the failure mode. 
As such, there is no final debug message(s) before a failure, and in fact when the problem occurs it seems to throw uMTP into a level of frantic activity.
With debug enabled and '--background' removed from umtprd.init, I ran "$ sudo /home/pi/uMTP-Responder/conf/umtprd.init start > /home/pi/log.txt " to record log files representing the issue.

I recorded 3 debug log files in this fashion, with the file sizes being ~10-17 MB. The smaller log files were created by stopping logging shortly after the problem appeared, where the larger log file was allowed to run for 10-30 seconds after the problem presented.

[log.txt](https://github.com/viveris/uMTP-Responder/files/7794480/log.txt)
[log_short.txt](https://github.com/viveris/uMTP-Responder/files/7794481/log_short.txt)
[log_short_2.txt](https://github.com/viveris/uMTP-Responder/files/7794482/log_short_2.txt)

Hopefully these will help to illustrate the circumstance.

jfdelnero, 2021-12-30:
Ok i see. Yes for some unknown reasons the Windows MTP client appears to want to scan all the store. I have already seen this behavior.
Are you changing/writing something though the mtp link to produce this behavior ?

jfdelnero, 2022-01-03:
After a closer look to the log.txt file it appears that you got a lot of inotify events from the /dev folders. This behavior appears to be ok to me. But i should maybe add an inotify folder "blacklist" feature to avoid/filter such useless events...

jfdelnero, 2022-01-03:
Another and better solution : add a delay (1 or 2 seconds) before sending the MTP event, and cancel or update it if we got another inotify event on the same object during this interval.

AndrewLloydSmith, 2022-01-03:
In order to trigger the issue I use Windows Explorer to navigate to the directory that the script running on the Pi is writing to. In this case that is /dev/shm/, though the same thing happens if I use another directory e.g. if I have a script running that modifies /home/pi/Documents and navigate there on the Windows machine.

Thank you for the suggestions, I will try them shortly.

jfdelnero, 2022-01-04:
Yes this is because your script is "visiting" folders containing files changing all the time (mostly in /dev). The umtpd daemon is then overflowed by inotify "modify" events. I need to change the way the inotify events are handled to avoid this.

AndrewLloydSmith, 2022-01-10:
Realizing that the issue stemmed from the mass scanning of /dev/ and other directories, I tried running uMTP with only the exact folder I wanted to read as an entry point. I setup umtprd.conf to only have:
storage "/dev/shm/"      "shm" "rw"

This seems to prevent the issue entirely, and allows me to run everything with inotify enabled. I will take more time to test this as a resolution for my specific case, and confirm later today. 

While of course this doesn't affect the circumstance of sharing root, I personally have no issue with the condition that entry points should be limited to the exact directory/directories of interest.

jfdelnero, 2022-01-10:
Thanks for the feedback. I will anyway modify the inotify layer to avoid this overflow situation in a next release.

closed 2023-06-08 by AndrewLloydSmith