viveris/uMTP-Responder #74: Spaces in Storage File Path Issue

mabrams17, 2022-02-14:
Hi,

When we have a space in the file path in the .conf file like either of these
`storage "/home/pi/Data Storage" "Data" "rw"`
`storage "/home/pi/'Data Storage'" "Data" "rw"`

It still connects to the machine and pops up as a flashdrive, but when you go in on windows you aren't able to add any files to the folder. 

It gives the following Warning & Error codes:
`Warning] Failed to get statvfs for /home/pi/'Data Storage'`
`Warning] MTP_OPERATION_GET_OBJECT_HANDLES : FOLDER ACCESS ERROR !`

As soon as we remove the space and just make it Data_Storage or DataStorage or something it works fine with no issues.

jfdelnero, 2022-03-15:
I tried but i didn't managed to reproduce this issue with the version 1.6.2.

`storage "/home/pi/Data Storage" "Data" "rw"`
works just fine here on a Raspberry Pi Zero.

Are you sure that the folder access permissions allow to umtprd to access to this folder ?

mabrams17, 2022-03-16:
The access permissions allow umtprd to access the folder. 

We were testing this on a Compute Module 4. I guess I should of tried this on different versions of the pi. After seeing you say it worked on the Pi Zero, I moved the sd card over to the Pi Zero (after re-adding the space & changing back to GadgetFS) and it worked fine. 

Wanted to find out more, so I tried both FunctionFS and GadgetFS on both the Compute Module and Pi Zero. Worked fine with the space for FunctionFS  on both. GadgetFS did not like the space and lead to the same Warning I originally posted when on the Compute Module although it did work on the Pi Zero.