viveris/uMTP-Responder #75: GadgetFS Looping does not reconnect when removed

mabrams17, 2022-02-15:
Hi,

On Compute Module 4, when you have a stable connection with a windows machine. If you remove the connection. Then plug it into a new pc or back into the same machine it will not reconnect.

I've traced it to in here `handle_ep0(usb_gadget * ctx)`. It seems to be going in an infinite loop on trying to reconnect with the timeout.tv_sec

mabrams17, 2022-02-15:
GadgetFS mode

mabrams17, 2022-02-16:
So finally got the debug information finally posting now. So I have more information to share.

While running and connected we get these messages

`handle_ep0 : Entering...`
`handle_ep0 : 1 event(s)`
`handle_ep0 : EP0 CONNECT EVENT`
`handle_ep0 : Select without timeout`

Then repeats below:
`handle_ep0 : 1 event(s)`
`handle_ep0 : EP0 SETUP event`
`handle_ep0 : Select without timeout`

When we disconnect we get the messages:
`handle_ep0 : 1 event(s)`
`handle_ep0 : EP0 SUSPEND event`
`handle_ep0 : Select without timeout`

`handle_ep0 : EP0 DISCONNECT event`
`handle_ep0 : Leaving (ctx_>stop=1)...`

Then we get these messages repeating
`uMTP Responder : Disconnected`
`handle_ep0 : Entering...`

Then when we plug in again we get these messages
`handle_ep0 : 1 event(s)`
`handle_ep0 : EP0 CONNECT event`
`handle_ep0 : Select without timeout`

Then a repeat of this message 6 times total
`handle_ep0 : 1 event(s)`
`handle_ep0 : EP0 SETUP event`
`handle_ep0 : Select without timeout`

Then after we get this message and it just stops
`handle_ep0 : 1 event(s)`
`handle_ep0 : EP0 SUSPEND event`
`handle_ep0 : Select without timeout`

jfdelnero, 2022-02-16:
Can you try the functionfs mode ?

mabrams17, 2022-02-16:
Trying with FunctionFS mode right now. Works perfectly. Can unplug and plug it in anywhere else and connects right away with no issues.