viveris/uMTP-Responder #76: Not working properly

Swapnil-Turambekar, 2022-03-09:
Hi,

I am using imx8m plus custom board. And I am trying to use Gadgetfs mode to access the the data from imx board using micro B USB. In Evk they used type C connector. It is working for some times only. can you help me here. I am attaching the failure logs for your reference.

root@imx8mp-lpddr4-evk:~# sh umtprd_gfs.sh 
[  558.197701] gadgetfs: USB Gadget filesystem, version 24 Aug 2004
[  558.223574] gadgetfs: bound to dwc3-gadget driver
[  558.499145] gadgetfs: connected
[  558.506630] gadgetfs: disconnected
[  558.516603] gadgetfs: bound to dwc3-gadget driver
[  558.651050] gadgetfs: connected
[  558.658396] gadgetfs: configuration #1
[  558.741025] gadgetfs: disconnected
[  558.750731] gadgetfs: bound to dwc3-gadget driver
[  558.863716] gadgetfs: connected
[  558.870857] gadgetfs: disconnected
[  558.880812] gadgetfs: bound to dwc3-gadget driver
[  559.014984] gadgetfs: connected
[  559.021254] gadgetfs: configuration #1
[  626.032060] gadgetfs: disconnected
[  626.035965] ------------[ cut here ]------------
[  626.040597] WARNING: CPU: 0 PID: 445 at drivers/usb/dwc3/gadget.c:3240 dwc3_stop_active_transfer.part.0.constprop.0+0xac/0xc0
[  626.051899] Modules linked in: gadgetfs fsl_jr_uio caam_jr caamkeyblob_desc caamhash_desc caamalg_desc crypto_engine rng_core a
uthenc libdes crct10dif_ce imx8_media_dev(C) secvio flexcan caam error can_dev fuse
[  626.070662] CPU: 0 PID: 445 Comm: irq/87-dwc3 Tainted: G         C        5.10.72-lts-5.10.y+ga68e31b63f86 #1
[  626.080574] Hardware name: NXP i.MX8MPlus EVK board (DT)
[  626.085888] pstate: 60000085 (nZCv daIf -PAN -UAO -TCO BTYPE=--)
[  626.091894] pc : dwc3_stop_active_transfer.part.0.constprop.0+0xac/0xc0
[  626.098510] lr : dwc3_stop_active_transfer.part.0.constprop.0+0x54/0xc0
[  626.105123] sp : ffff800012483c60
[  626.108438] x29: ffff800012483c60 x28: ffff0000c19af080 
[  626.113753] x27: ffff0000c6559d48 x26: ffff0000c1a83080 
[  626.119069] x25: 0000000000000000 x24: 0000000000000001 
[  626.124385] x23: ffff8000116c55b8 x22: 0000000000000000 
[  626.129703] x21: 0000000000000000 x20: ffff0000c735aac0 
[  626.135021] x19: ffff0000c19af700 x18: ffff800011b31788 
[  626.140339] x17: 0000000000000000 x16: 0000000000000000 
[  626.145655] x15: 0000000000000020 x14: 00000000000002b6 
[  626.150971] x13: 0000000000000000 x12: 0000000000000040 
[  626.156287] x11: ffff0000c2000000 x10: ffff0000c2000002 
[  626.161606] x9 : ffff800011b2f228 x8 : ffff0000c2000028 
[  626.166922] x7 : 0000000000000000 x6 : 0000000000030c08 
[  626.172240] x5 : ffffffffffff3f0c x4 : 0000000000000000 
[  626.177558] x3 : ffffffffffff3f08 x2 : 0000000000000000 
[  626.182876] x1 : ffff800013f84200 x0 : 00000000ffffff92 
[  626.188192] Call trace:
[  626.190644]  dwc3_stop_active_transfer.part.0.constprop.0+0xac/0xc0
[  626.196912]  dwc3_remove_requests.constprop.0+0x2c/0xf0
[  626.202142]  dwc3_gadget_reset_interrupt+0x70/0x190
[  626.207019]  dwc3_thread_interrupt+0x354/0xad0
[  626.211466]  irq_thread_fn+0x2c/0xa0
[  626.215041]  irq_thread+0x160/0x280
[  626.218532]  kthread+0x154/0x160
[  626.221763]  ret_from_fork+0x10/0x30
[  626.225339] ---[ end trace b3e367c1072be79d ]---
[  626.752027] dwc3 38100000.dwc3: timed out waiting for SETUP phase
[  626.765109] gadgetfs: bound to dwc3-gadget driver
[  626.770440] dwc3 38100000.dwc3: failed to enable ep0out
[  631.296024] dwc3 38100000.dwc3: timed out waiting for SETUP phase
[  631.307803] gadgetfs: bound to dwc3-gadget driver
[  631.313133] dwc3 38100000.dwc3: failed to enable ep0out
^C^C[  635.488024] dwc3 38100000.dwc3: timed out waiting for SETUP phase

kstarfss, 2022-05-17:
Hi,
I am seeing this issue on 1.3.10 and 1.6.2 versions.  Hardware is Xilinx UltraScale+ and I am using FunctionFs mode on Linux kernel 5.4.0:
_Linux PLMOS 5.4.0-xilinx-v2020.1 #1 SMP Fri May 13 22:04:49 UTC 2022 aarch64 aarch64 aarch64 GNU/Linu_

Whenever umtprd is exited, I get this crash. I have seen it on 1.3.10 version.  I upgraded to 1.6.2 version hoping it might resolve the issue and now it doesn't enumerate on bootup.  If I manually try starting umtprd from the command line, it starts ok, but it does not enumerate on the Windows side with an invalid device descriptor error.  On 1.3.10 it enumerates ok, but on both versions I see this crash when umtprd is exited.  Below is console output with debug on and dmesg output:

May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] uMTP Responder
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Version: v1.6.2 compiled the May 16 2022@22:55:11
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] (c) 2018 - 2021 Viveris Technologies
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] init_mtp_responder
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] init_inotify_handler : inotify_fd = 4
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] msgqueue_handler_init : current exec path : /usr/bin/umtprd
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] msgqueue_handler_init : msgqueue_id = 7
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] init_mtp_responder : Ok !
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Add storage PLM Storage - Root Path: /mnt/flash - UID: -1 - GID: -1 - Flags: 0x00000000
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] mtp_add_storage : /mnt/flash
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] mtp_get_storage_id_by_name : PLM Storage
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] mtp_add_storage : Storage FFFF0001 mapped to /mnt/flash (PLM Storage) (Flags: 0x00000000)
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Device path : /dev/ffs-mtp/ep0
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB In End point path : /dev/ffs-mtp/ep1
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Out End point path : /dev/ffs-mtp/ep2
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Event End point path : /dev/ffs-mtp/ep3
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Max packet size : 0x200 bytes
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Max write buffer size : 0x2000 bytes
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Max read buffer size : 0x2000 bytes
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Read file buffer size : 0x100000 bytes
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Manufacturer string : Foresight Sports
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Product string : Personal Launch Monitor
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Serial string : 1000028
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Firmware Version string : Rev A
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Interface string : MTP
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Vendor ID : 0x2C79
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Product ID : 0x0140
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB class ID : 0x06
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB subclass ID : 0x01
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Protocol ID : 0x01
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB Device version : 0x3008
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] USB FunctionFS Mode
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Wait for connection : 1
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Loop on disconnect : 1
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Show hidden files : 1
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] File creation umask : System default umask
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Default UID : 0
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] Default GID : 0
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] inotify : yes
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] fill_if_descriptor:
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] 09 04 00 00 03 06 01 01 01                       .........        
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] fill_ep_descriptor:
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] 07 05 81 02 00 02 00                             .......          
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] fill_ep_descriptor:
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] 07 05 02 02 00 02 00                             .......          
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] fill_ep_descriptor:
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] 07 05 83 03 1C 00 06                             .......          
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] fill_if_descriptor:
May 16 16:05:00 PLMOS user.notice umtprd: [uMTPrd - Info] 09 04 00 00 03 06 01 01 01                       .........        
May 16 16:05:00 PLMOS user.debug umtprd: [uMTPrd - Debug] fill_ep_descriptor:
[... cut ...]
[  401.429913]  ffs_data_reset+0x14/0x48 [usb_f_fs]
[  401.434522]  ffs_data_closed+0xd4/0x130 [usb_f_fs]
[  401.439305]  ffs_ep0_release+0x10/0x20 [usb_f_fs]
[  401.444006]  __fput+0x88/0x208
[  401.447047]  ____fput+0xc/0x18
[  401.450088]  task_work_run+0xc8/0xf8
[  401.453654]  do_exit+0x2ac/0x9b8
[  401.456873]  do_group_exit+0x40/0xa8
[  401.460433]  get_signal+0x12c/0x720
[  401.463915]  do_notify_resume+0x18c/0x388
[  401.467914]  work_pending+0x8/0x10
[  401.471298] ---[ end trace 607d30ab5a3ff0a7 ]---

jfdelnero, 2022-05-17:
I see that you are using dwc3. Have you enabled "CONFIG_USB_SS_SUPPORT" in "buildconf.h" ?

jfdelnero, 2022-05-17:
If this still doesn't work with the CONFIG_USB_SS_SUPPORT enabled, try to disable the inotify support in the umtp config file.

jfdelnero, 2026-11-01:
Some SS support fixes are now on the repository.

jfdelnero, 2026-11-02:
Reopen if you still have the issue.

closed 2026-11-02 by jfdelnero