viveris/uMTP-Responder #77: Debian 64-bit Question

mabrams17, 2022-03-23:
Has this been updated to work on Debian Bullseye 64-bit?

I tried setting it up the same way I have done for Bullseye 32bit and it works on 32-bit but when I try it on 64-bit it doesn't connect in the first place but still repeatedly gives the message below as if it connected in the first place.
`[uMTPrd - 18:00:37 - Info] uMTP Responder : Disconnected`

jfdelnero, 2022-03-28:
In FunctionFs or Gadget mode ?

mabrams17, 2022-03-29:
FunctionFS.

jfdelnero, 2022-03-29:
Please recompile it with the debug output with CFLAGS="-DUSE_SYSLOG -DDEBUG" and post here the logs.

If possible also try to force the compilation in 32bits on the 64bits debian with the gcc -m32 option. You may need to install multilib (sudo apt-get install gcc-multilib). If umtprd works this way this means that i have to fix a structure/variable type.

colejohnson66, 2022-06-15:
On Debian Bullseye, compiled with `-DUSE_SYSLOG -DDEBUG`, we only get this in `/var/log/messages`:

```
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] uMTP Responder
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Version: v1.6.2 compiled the Jun 15 2022@10:46:17
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] (c) 2018 - 2021 Viveris Technologies
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Add storage www folder - Root Path: /www - UID: -1 - GID: -1 - Flags: 0x00000003
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Add storage User Data - Root Path: /home/pi/<HIDDEN>/User_Data - UID: -1 - GID: -1 - Flags: 0x00000000
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Add storage Logs - Root Path: /home/pi/<HIDDEN>/Logs - UID: -1 - GID: -1 - Flags: 0x00000001
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Add storage lockable root folder - Root Path: / - UID: -1 - GID: -1 - Flags: 0x00000018
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Add storage user folder - Root Path: /home/user - UID: 33 - GID: 33 - Flags: 0x00000018
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Device path : /dev/ffs-mtp/ep0
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB In End point path : /dev/ffs-mtp/ep1
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Out End point path : /dev/ffs-mtp/ep2
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Event End point path : /dev/ffs-mtp/ep3
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Max packet size : 0x200 bytes
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Max write buffer size : 0x2000 bytes
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Max read buffer size : 0x2000 bytes
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Read file buffer size : 0x100000 bytes
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Manufacturer string : <HIDDEN>
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Product string : <HIDDEN>
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Serial string : 01234567
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Firmware Version string : Rev A
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Interface string : MTP
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Vendor ID : 0x1D6B
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Product ID : 0x0100
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB class ID : 0x06
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB subclass ID : 0x01
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Protocol ID : 0x01
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB Device version : 0x3008
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] USB FunctionFS Mode
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Wait for connection : 0
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Loop on disconnect : 1
Jun 15 11:41:02 raspberrypi umtprd: [uMTPrd - Info] Show hidden files : 1
Jun 15 11:41:03 raspberrypi umtprd: [uMTPrd - Info] File creation umask : System default umask
Jun 15 11:41:03 raspberrypi umtprd: [uMTPrd - Info] Default UID : 1000
Jun 15 11:41:03 raspberrypi umtprd: [uMTPrd - Info] Default GID : 1000
Jun 15 11:41:03 raspberrypi umtprd: [uMTPrd - Info] inotify : yes
Jun 15 11:41:03 raspberrypi umtprd: [uMTPrd - Info] uMTP Responder : Disconnected
```

```
$ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

colejohnson66, 2022-06-15:
Raspbian also does not include a `gcc-multilib` package for some reason, so I'm unable to test compiling a 32 bit binary on a 64 bit system.