viveris/uMTP-Responder #92: Windows 10 can cause segment errors

1Keeptrying, 2023-09-14:
The built-in player in Windows 10 can cause segment errors when playing videos on the device, causing the device to get stuck. In addition, canceling the transmission of files during transmission can also cause segment errors. Currently, I am using the latest code. Can these issues be resolved? thanks

jfdelnero, 2023-11-07:
if possible please build it with the debug infos. some logs are needed.
i also need to know the setup you are using : target system and kernel version.

DarrenHello, 2024-01-15:
First of all, thank you for this great project!  I'm having the same issue.The built-in player in Windows 10/11  seems to cause the device to get stuck when playing videos on PC. I find that the video file can be played once,but need long time . when I play it again, the playback is interrupted and an error code 0x800700aa is reported（mtp device is busy）.Unplug the USB and plug it back in，I can't discover the MTP device on my PC. Reboot  the device to resolve the issue. I don't encounter the above problem after changing other players,such as vlc.

DarrenHello, 2024-01-16:
[image]
[image]
[image]
[image]
[image]

jfdelnero, 2026-11-01:
I have tested this on a Samsung Galaxy Tab Active5 5G with Android 16 and Windows 11 to compare. The behavior is exactly the same, so the issue is not related to uMTP-Responder but to the Windows MTP initiator.

closed 2026-11-01 by jfdelnero