viveris/uMTP-Responder #99: Wrong name in file/folder creation

tiopex, 2024-01-02:
While creating a folder it returns Chinese characters. It only works correctly when moving/copying files from outside.

It happens on Windows 10 and 11.

below are debug logs from create "test" directory. There is "test" in the payload but in "build_full_path" unexpected Chinese characters.

```
[uMTPrd - 11:07:13 - Debug] --------------------------------------------------
[uMTPrd - 11:07:13 - Debug] Incoming_packet : 0x48c5e8 - rawsize : 20
[uMTPrd - 11:07:13 - Debug] MTP Packet size : 20 bytes
[uMTPrd - 11:07:13 - Debug] MTP Operation   : 0x0001 (OPERATION)
[uMTPrd - 11:07:13 - Debug] MTP code        : 0x9804 (MTP_OPERATION_SET_OBJECT_PROP_VALUE)
[uMTPrd - 11:07:13 - Debug] MTP Tx ID       : 0x00000078
[uMTPrd - 11:07:13 - Debug] Header : 
[uMTPrd - 11:07:13 - Debug] 14 00 00 00 01 00 04 98 78 00 00 00              ........x...     
[uMTPrd - 11:07:13 - Debug] Payload : 
[uMTPrd - 11:07:13 - Debug] 12 00 00 00 07 DC 00 00                          ........         
[uMTPrd - 11:07:13 - Debug] No Status response sent
[uMTPrd - 11:07:13 - Debug] --------------------------------------------------
[uMTPrd - 11:07:13 - Debug] Incoming_packet : 0x48c5e8 - rawsize : 23
[uMTPrd - 11:07:13 - Debug] MTP Packet size : 23 bytes
[uMTPrd - 11:07:13 - Debug] MTP Operation   : 0x0002 (DATA)
[uMTPrd - 11:07:13 - Debug] MTP code        : 0x9804 (MTP_OPERATION_SET_OBJECT_PROP_VALUE)
[uMTPrd - 11:07:13 - Debug] MTP Tx ID       : 0x00000078
[uMTPrd - 11:07:13 - Debug] Header : 
[uMTPrd - 11:07:13 - Debug] 17 00 00 00 02 00 04 98 78 00 00 00              ........x...     
[uMTPrd - 11:07:13 - Debug] Payload : 
[uMTPrd - 11:07:13 - Debug] 05 74 00 65 00 73 00 74 00 00 00                 .t.e.s.t...      
[uMTPrd - 11:07:13 - Debug] setObjectPropValue : (Handle : 0x00000012 - Prop code : 0xDC07 )
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_flags : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_flags : FFFF0002 -> 0x00000000
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[uMTPrd - 11:07:13 - Debug] build_full_path : Nowy folder -> /mnt/Nowy folder
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[uMTPrd - 11:07:13 - Debug] build_full_path : 琅攀猀琀 -> /mnt/琅攀猀琀
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_index_by_id : 0xFFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_index_by_id : FFFF0002 -> 1
[uMTPrd - 11:07:13 - Debug] Entry 0 - /
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[uMTPrd - 11:07:13 - Debug] build_full_path : / -> /mnt//
[uMTPrd - 11:07:13 - Debug] inotify_thread (IN_DELETE): Watch point descriptor not found in the db ! (Descriptor 0x00000001)
[uMTPrd - 11:07:13 - Debug] Entry 0 - /
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[uMTPrd - 11:07:13 - Debug] build_full_path : / -> /mnt//
[uMTPrd - 11:07:13 - Debug] inotify_thread (IN_CREATE): Entry 琅攀猀琀 already in the db ! (Handle 0x00000012)
[uMTPrd - 11:07:13 - Debug] inotify_thread (IN_CREATE): Watch point descriptor not found in the db ! (Descriptor 0x00000001)
[uMTPrd - 11:07:13 - Debug] Status response (12 Bytes):
[uMTPrd - 11:07:13 - Debug] 0C 00 00 00 03 00 01 20 78 00 00 00              ........x...     
[uMTPrd - 11:07:13 - Debug] --------------------------------------------------
[uMTPrd - 11:07:13 - Debug] Incoming_packet : 0x48c5e8 - rawsize : 32
[uMTPrd - 11:07:13 - Debug] MTP Packet size : 32 bytes
[uMTPrd - 11:07:13 - Debug] MTP Operation   : 0x0001 (OPERATION)
[uMTPrd - 11:07:13 - Debug] MTP code        : 0x9805 (MTP_OPERATION_GET_OBJECT_PROP_LIST )
[uMTPrd - 11:07:13 - Debug] MTP Tx ID       : 0x00000079
[uMTPrd - 11:07:13 - Debug] Header : 
[uMTPrd - 11:07:13 - Debug] 20 00 00 00 01 00 05 98 79 00 00 00              ........y...     
[uMTPrd - 11:07:13 - Debug] Payload : 
[uMTPrd - 11:07:13 - Debug] 12 00 00 00 00 00 00 00 02 DC 00 00 00 00 00 00  ................ 
[uMTPrd - 11:07:13 - Debug] 00 00 00 00                                      ....             
[uMTPrd - 11:07:13 - Debug] MTP_OPERATION_GET_OBJECT_PROP_LIST :(Handle: 0x00000012 FormatCode: 0x00000000 ObjPropCode: 0x0000DC02 ObjPropGroupCode: 0x00000000 Depth: 0)
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[... 47 lines cut ...]
[uMTPrd - 11:07:13 - Debug] Header : 
[uMTPrd - 11:07:13 - Debug] 10 00 00 00 01 00 08 10 7B 00 00 00              ........{...     
[uMTPrd - 11:07:13 - Debug] Payload : 
[uMTPrd - 11:07:13 - Debug] 12 00 00 00                                      ....             
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:13 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
[uMTPrd - 11:07:13 - Debug] build_full_path : 琅攀猀琀 -> /mnt/琅攀猀琀
[uMTPrd - 11:07:13 - Debug] MTP_OPERATION_GET_OBJECT_INFO response (142 Bytes):
[uMTPrd - 11:07:13 - Debug] 8E 00 00 00 02 00 08 10 7B 00 00 00 02 00 FF FF  ........{....... 
[uMTPrd - 11:07:13 - Debug] 01 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00  .0.............. 
[uMTPrd - 11:07:13 - Debug] 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................ 
[uMTPrd - 11:07:13 - Debug] 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00  ................ 
[uMTPrd - 11:07:13 - Debug] 05 05 74 00 65 00 73 00 74 00 00 10 32 00 30 00  ..t.e.s.t...2.0. 
[uMTPrd - 11:07:13 - Debug] 32 00 33 00 31 00 32 00 32 00 39 00 54 00 31 00  2.3.1.2.2.9.T.1. 
[uMTPrd - 11:07:13 - Debug] 31 00 30 00 37 00 30 00 38 00 00 00 10 32 00 30  1.0.7.0.8....2.0 
[uMTPrd - 11:07:13 - Debug] 00 32 00 33 00 31 00 32 00 32 00 39 00 54 00 31  .2.3.1.2.2.9.T.1 
[uMTPrd - 11:07:13 - Debug] 00 31 00 30 00 37 00 30 00 38 00 00 00 00        .1.0.7.0.8....   
[uMTPrd - 11:07:13 - Debug] Status response (12 Bytes):
[uMTPrd - 11:07:13 - Debug] 0C 00 00 00 03 00 01 20 7B 00 00 00              ........{...     
[uMTPrd - 11:07:18 - Debug] Entry 0 - /
[uMTPrd - 11:07:18 - Debug] mtp_get_storage_root : FFFF0002
[uMTPrd - 11:07:18 - Debug] mtp_get_storage_root : FFFF0002 -> /mnt
```

jfdelnero, 2024-01-02:
Hello,
On which CPU / machine is running the uMTP daemon ? Is it a big-endian system ?
And which Linux it is ?

tiopex, 2024-01-02:
> Hello, On which CPU / machine is running the uMTP daemon ? Is it a big-endian system ? And which Linux it is ?

ARM926EJ-S (ARMv5TE) little-endian, kernel 5.15, buildroot 2022.02.9

jfdelnero, 2024-01-02:
Ok i see what probably cause the issue : ARMv5 doesn't support unaligned memory accesses... unicode2charstring use short accesses to read  the unicode words, and high and low bytes are swapped on this CPU and generate these bad characters. I will change this function to fix this.

jfdelnero, 2024-01-02:
@tiopex : I have pushed a possible fix on the master branch. Please try it and let me know if this fix the issue.

tiopex, 2024-01-03:
Works! Thank you.

closed 2024-01-03 by tiopex