viveris/uMTP-Responder #102: inotify not working

9WW7XxHy, 2024-03-28:
when i change the folder inside the mtp side doesn't seem to update

marcone, 2024-11-07:
I noticed this too (with Windows 10), and when I looked in to it it seems that mtprd does notify Windows that something changed, and Windows then responds by requesting information about the change (e.g. when I create a new file, Windows responds with MTP_OPERATION_GET_OBJECT_PROP_DESC and MTP_OPERATION_GET_OBJECT_PROP_LIST), but then doesn't actually update the Explorer window.

jfdelnero, 2026-11-01:
I have seen the same issue in the past, but I suspect that this is a Windows MTP initiator issue again. Refreshing the Explorer page with F5 updates the folder content.

I will try this with an Android machine to compare.