viveris/uMTP-Responder #103: New release soon?

JamiKettunen, 2024-04-27:
There has been a bunch of [nice fixes since 1.6.2](https://github.com/viveris/uMTP-Responder/compare/umtprd-1.6.2...master) and the commit activity has started to slow down, would it make sense to cut a new release again after 2 years?

JamiKettunen, 2024-06-05:
Ping @jfdelnero, thanks for the work on this project btw

jfdelnero, 2024-06-14:
Yes i should release a new version. 
Are you currently testing / using the repository version ?

JamiKettunen, 2024-06-17:
Yes, it works nicely on a modern Android device with Linux kernel v5.10 with an alternative userspace (Ubuntu Touch) where it may become the new MTP backend with the upcoming releases based on 24.04 :)

jfdelnero, 2024-06-20:
Interesting. 
Are you testing it with different MTP clients ? (Windows 10 / 11, macOS, Linux, Android...)
 
How will be managed / designed the unlock user interface ?

JamiKettunen, 2024-06-21:
The [Ubuntu Touch](https://ubuntu-touch.io) integration with https://github.com/sailfishos/usb-moded (for USB gadget mode control) looks [something like this for example](https://gitlab.com/ubports/porting/community-ports/android12/volla-tablet/volla-mimir/-/compare/15ed7335...17b753bb). I've mostly only used it with Linux as a client but reportedly something after v1.6.2 has also fixed it for macOS (in terms of copying files) too as tested on 7d7d099.

On my end [KDE's `kio-extras` MTP impl](https://invent.kde.org/network/kio-extras/-/tree/master/mtp) also became usable, though it's still somewhat finnicky with the unlocking not making internal storage shown in case device was still locked while plugging in, but I'd put this down to a broken implementation since with [GNOME's `gvfs`](https://gitlab.gnome.org/GNOME/gvfs/-/blob/master/daemon/gvfsbackendmtp.c) consumers it appears works fine.

From a quick test Windows 10 as a client also works including plugging in before device unlock (possibly after following https://www.droidviews.com/fix-mtp-driver-installation-on-windows-10/? not sure if I already did this before), I don't have 11 handy anywhere myself but maybe it could be tested in a VM until I have my new laptop replacement anyway (~2 weeks max hopefully).

jfdelnero, 2024-07-08:
i have finally released the v1.6.8 :
https://github.com/viveris/uMTP-Responder/releases/tag/umtprd-1.6.8

JamiKettunen, 2024-07-08:
Thank you! Closing this now :)

closed 2024-07-08 by JamiKettunen