viveris/uMTP-Responder #108: mp4 files cannot be played directly, need to be copied first.

marcone, 2024-09-12:
I set this up on a Raspberry Pi Zero W and pointed it at a folder full of mp4 files. The files show up on the attached Windows 10 PC, and Windows by default opens them with Media Player, however they won't play (Media Player just shows the progress spinner animation forever).
If I copy (using drag&drop) them to the Windows hard drive first, the files can be played from there.

jfdelnero, 2024-11-03:
Can you make a wireshark usb dump ?
Have you another device/smartphone to compare ?

marcone, 2024-11-03:
I'll look into wireshark. For now, I've attached umtprd debug logs showing the process of navigating to the folder where the file is located, then double clicking on the file to open it with Windows Media Player.
After this, Media Player hangs indefinitely, and after closing Media Player, I can no longer access the MTP device at all (any operation just makes Windows block), and have to restart umtprd.

[umtprdlog.txt](https://github.com/user-attachments/files/17611407/umtprdlog.txt)

(note that this logging is with https://github.com/viveris/uMTP-Responder/pull/112 applied, so it'll look slightly different than what you're used to)

I also tried without debug logs, in case the problem is related to logging, or because enabling debug logs also disables compiler optimizations, but it behaves the same with and without logs/debug mode.

These logs are from a Raspberry Pi Zero **2** W, not the Raspberry Pi Zero W that I originally tried.

marcone, 2024-11-03:
Here's a Wireshark capture. Not sure I did it right, since it seems a bit short.
What I did was uncheck the "Capture from all devices connected" checkbox in the USBPcap1 settings (since I also have a mouse, keyboard, drives, etc. and I didn't want it to capture those), then start the capture, then start umtprd so Wireshark saw it as a new device and would start capturing it. I then opened the MTP device, navigated to a short (748KB) test video, and double clicked it to open Media Player. I let that spin for a while, then closed Media Player, stopped the capture, and saved the attached capture file.

[mtpcapture.pcapng.gz](https://github.com/user-attachments/files/17611625/mtpcapture.pcapng.gz)

marcone, 2024-11-03:
For what's worth, I also tried using gadgetfs instead of functionfs, which behaved the same when trying to play a file, but caused the Pi to go into an infinite loop and become unresponsive when stopping umtrpd.

marcone, 2024-11-04:
Also, to make sure the problem isn't with Windows or Media Player I copied the same test video to my Pixel 5, and verified that Windows/Media Player could play the file just fine when accessing the file using MTP from the Pixel 5.

marcone, 2024-11-04:
Here's a capture of the Pixel 5. Started Wireshark, then enabled the "File transfer / Android Auto" USB mode, then opened the "Pixel 5" device in Windows and played the test file.

[mtpcapturepixel.pcapng.gz](https://github.com/user-attachments/files/17623009/mtpcapturepixel.pcapng.gz)

marcone, 2024-11-04:
> Anyway this is for sure related to the Setup Request 0x64 MTP_REQ_CANCEL operation sent by the PC

That request doesn't happen until Media Player has already been spinning for 30 seconds or so though, so that can't be the reason why it doesn't start playback.

marcone, 2024-11-06:
> > Anyway this is for sure related to the Setup Request 0x64 MTP_REQ_CANCEL operation sent by the PC
> 
> That request doesn't happen until Media Player has already been spinning for 30 seconds or so though, so that can't be the reason why it doesn't start playback.

Correction: it might be the MTP_CANCEL_REQ that is the problem, however it doesn't seem like it is caused by the URB_FUNCTION_ABORT_PIPE from the host, since I don't see that ...ABORT_PIPE until it already hangs.

I looked in to this some more, and it appears the problem is that [this call to write_usb](https://github.com/marcone/uMTP-Responder/blob/master/src/mtp_operations/mtp_ops_helpers.c#L155) blocks forever. I tried enabling CONFIG_USB_NON_BLOCKING_WRITE but that didn't help (it still blocks on the call to `write()` inside of `write_usb()`, even though both the `fnctl()` to make it nonblocking as well as the `poll()` to check whether it can receive data succeed).

jfdelnero, 2024-11-07:
Same issue : 
https://github.com/viveris/uMTP-Responder/issues/92

jfdelnero, 2024-11-07:
Another one :
https://github.com/viveris/uMTP-Responder/issues/30

jfdelnero, 2024-11-07:
This one is interesting :
https://github.com/viveris/uMTP-Responder/issues/96

marcone, 2024-11-07:
Those other issues all seem to talk about kernel crashes/segfaults, which I'm not seeing. I'm just seeing that `write()` in `write_usb()` hang forever, unless I remove the `read()` in the `MTP_REQ_CANCEL` case.

jfdelnero, 2024-11-07:
can you try to cancel a big file transfer (from and to the device) ?

marcone, 2024-11-07:
Copying a large file from Windows to the device I noticed the following:
- the Windows progress indicator is all wrong. It slowly goes to about 20% or so, and then the transfer is complete. 
- canceling the copy works fine, with no crashes in either umtprd or the kernel

Copying a large file from the device to Windows:
- the Windows progress indicator appears correct
- letting the file copy without interruption works fine
- canceling the transfer hangs, and required restarting umtprd afterwards
- no crashes observed in either umtprd or the kernel

If I remove the `read()` call like I did in https://github.com/viveris/uMTP-Responder/commit/1e6936a432442a0fddb1ecfa178dd6e97d322434 then canceling the file transfer to the device works OK.

jfdelnero, 2024-11-07:
ok ! 
I hope that this is not device / soc dependent..

closed 2024-11-07 by jfdelnero

marcone, 2024-11-08:
> ok ! I hope that this is not device / soc dependent..

It looks like it is device dependent. I just tried this on a Rock Pi 4C+ running Armbian with kernel 6.1.63, and removing the `read()` doesn't appear to be necessary there (but removing it doesn't seem to cause any problems either)

However, also on the Rock Pi 4C+, I could only get it to work if I connect the device to the PC as a USB2 device. When using it as a USB3 device, mtprd starts up seemingly fine and there are no errors in umtprd's debug log or dmesg, but the PC complains about "USB device not recognized", while USB Device Tree Viewer says there's a port malfunction.

Unrelated to this, but I also found that Windows apparently caches the product name of the device based on the serial number. Once it has seen device with a serial number 'S' and name 'N', any subsequent device that uses serial number 'S' will be shown with name 'N' in Explorer, even if a different name is used in strings/0x409/product. Right-clicking on the device and selecting 'properties' shows both the cached name 'N' and the actual name from the product string.

commit 1e6936a4 (2024-11-07): Fix video playback on Windows 10

Doing a blocking read (even with count=0) when handling a cancel request
from the host ends up blocking both the read in the request handler as
well as a write in the thread that's being cancelled, blocking both threads
indefinitely.
Removing the read appears to fix the problem.
Fixes #108