viveris/uMTP-Responder #126: Unable to send files

tiopex, 2025-01-24:
I'm using Allwinner H700 board with 6.12 kernel. During sending files to board, communication hangs with below logs

Umtprd:
```
[uMTPrd - 02:57:41 - Debug] Incoming_packet : 0x15dc090 - rawsize : 0
[uMTPrd - 02:57:44 - Debug] 1 event(s)
[uMTPrd - 02:57:44 - Debug] EP0 FFS SETUP
[uMTPrd - 02:57:44 - Debug] Setup request 0x67
[uMTPrd - 02:57:44 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:57:44 - Debug] Select without timeout
[uMTPrd - 02:57:47 - Debug] 1 event(s)
[uMTPrd - 02:57:47 - Debug] EP0 FFS SETUP
[uMTPrd - 02:57:47 - Debug] Setup request 0x67
[uMTPrd - 02:57:47 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:57:47 - Debug] Select without timeout
[uMTPrd - 02:57:50 - Debug] 1 event(s)
[uMTPrd - 02:57:50 - Debug] EP0 FFS SETUP
[uMTPrd - 02:57:50 - Debug] Setup request 0x67
[uMTPrd - 02:57:50 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:57:50 - Debug] Select without timeout
[uMTPrd - 02:57:53 - Debug] 1 event(s)
[uMTPrd - 02:57:53 - Debug] EP0 FFS SETUP
[uMTPrd - 02:57:53 - Debug] Setup request 0x67
[uMTPrd - 02:57:53 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:57:53 - Debug] Select without timeout
[uMTPrd - 02:57:56 - Debug] 1 event(s)
[uMTPrd - 02:57:56 - Debug] EP0 FFS SETUP
[uMTPrd - 02:57:56 - Debug] Setup request 0x67
[uMTPrd - 02:57:56 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:57:56 - Debug] Select without timeout
[uMTPrd - 02:58:00 - Debug] 1 event(s)
[uMTPrd - 02:58:00 - Debug] EP0 FFS SETUP
[uMTPrd - 02:58:00 - Debug] Setup request 0x67
[uMTPrd - 02:58:00 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:58:00 - Debug] Select without timeout
[uMTPrd - 02:58:03 - Debug] 1 event(s)
[uMTPrd - 02:58:03 - Debug] EP0 FFS SETUP
[uMTPrd - 02:58:03 - Debug] Setup request 0x67
[uMTPrd - 02:58:03 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:58:03 - Debug] Select without timeout
[uMTPrd - 02:58:06 - Debug] 1 event(s)
[uMTPrd - 02:58:06 - Debug] EP0 FFS SETUP
[uMTPrd - 02:58:06 - Debug] Setup request 0x67
[uMTPrd - 02:58:06 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:58:06 - Debug] Select without timeout
[uMTPrd - 02:58:09 - Debug] 1 event(s)
[uMTPrd - 02:58:09 - Debug] EP0 FFS SETUP
[uMTPrd - 02:58:09 - Debug] Setup request 0x67
[uMTPrd - 02:58:09 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:58:09 - Debug] Select without timeout
[uMTPrd - 02:58:12 - Debug] 1 event(s)
[uMTPrd - 02:58:12 - Debug] EP0 FFS SETUP
[uMTPrd - 02:58:12 - Debug] Setup request 0x67
[uMTPrd - 02:58:12 - Debug] MTP_REQ_GET_DEVICE_STATUS !
[uMTPrd - 02:58:12 - Debug] Select without timeout
[uMTPrd - 02:58:12 - Debug] 1 event(s)
[uMTPrd - 02:58:12 - Debug] EP0 FFS SETUP
[uMTPrd - 02:58:12 - Debug] Setup request 0x64
[uMTPrd - 02:58:12 - Debug] MTP_REQ_CANCEL !
[uMTPrd - 02:58:12 - Debug] MTP_REQ_CANCEL : Forcing read/write exit...
[uMTPrd - 02:58:14 - Debug] MTP_REQ_CANCEL : Stalled ... Killing the link...
[uMTPrd - 02:58:14 - Debug] MTP_REQ_CANCEL done !
[uMTPrd - 02:58:14 - Debug] handle_ffs_ep0 : Leaving... (ctx->stop=1)
[uMTPrd - 02:58:14 - Debug] entering deinit_usb_mtp_gadget
[uMTPrd - 02:58:14 - Debug] Closing End Point 0...
[uMTPrd - 02:58:14 - Debug] Closing End Point 1...
[uMTPrd - 02:58:14 - Debug] Closing End Point 2...
[uMTPrd - 02:58:14 - Debug] Closing usb device...
[uMTPrd - 02:58:14 - Debug] Stopping USB Thread...
[uMTPrd - 02:58:14 - Debug] leaving deinit_usb_mtp_gadget
[... 142 lines cut ...]
[  183.458001] : 04 00 01 20                                      ...
[  186.468230] configfs-gadget.mtp gadget.0: non-core control reqa1.67 v0000 i0000 l36
[  186.468269] creq->bRequestType = a1
[  186.468274] creq->bRequest     = 67
[  186.468278] creq->wValue       = 0000
[  186.468282] creq->wIndex       = 0000
[  186.468287] creq->wLength      = 0024
[  186.468292] adding event 4
[  186.468633] Buffer from user space:
[  186.468643] : 04 00 01 20                                      ...
[  189.472557] configfs-gadget.mtp gadget.0: non-core control reqa1.67 v0000 i0000 l36
[  189.472595] creq->bRequestType = a1
[  189.472600] creq->bRequest     = 67
[  189.472604] creq->wValue       = 0000
[  189.472609] creq->wIndex       = 0000
[  189.472613] creq->wLength      = 0024
[  189.472619] adding event 4
[  189.472926] Buffer from user space:
[  189.472936] : 04 00 01 20                                      ...
[  190.877415] udc musb-hdrc.2.auto: unbinding gadget driver [mtp]
[  190.877471] configfs-gadget.mtp gadget.0: reset config
[  190.877498] adding event 3
[  190.877518] configfs-gadget.mtp gadget.0: unbind function 'Function FS Gadget'/00000000c1fc48c9
[  190.877536] purging event 3
```

tiopex, 2025-01-27:
When compiled without the DEBUG flag, the transfer sometimes works but remains unstable. Additionally, setting usb_max_packet_size to 0x40 helps, but the transfer is still not stable. 

When the DEBUG flag is enabled, file transfers never succeed.

jfdelnero, 2025-01-28:
Have you the same issue with the last tagged version ?

https://github.com/viveris/uMTP-Responder/archive/refs/tags/umtprd-1.6.8.tar.gz

tiopex, 2025-01-30:
Yes, tried with last tagged and trunk

paulsykes, 2025-04-09:
@tiopex maybe related to this https://github.com/viveris/uMTP-Responder/issues/130

tiopex, 2025-04-10:
indeed, reverting https://github.com/gregkh/linux/commit/3fc137386c4620305bbc2a216868c53f9245670a fix the issue

paulsykes, 2025-04-10:
I looked into this a little bit more today. I think whilst reverting the commit appears to fix the issue. The kernel commit is correct and the previous kernel behaviour is masking an I/O issue in uMTP-Responder.  I need to investigate further to be able to confirm this.

tiopex, 2025-04-10:
I tried also https://github.com/cmtp-responder/cmtp-responder and behaviour was identical

paulsykes, 2025-04-11:
> I tried also https://github.com/cmtp-responder/cmtp-responder and behaviour was identical

@tiopex please can you clarify. Does https://github.com/cmtp-responder/cmtp-responder work before applying https://github.com/gregkh/linux/commit/3fc137386c4620305bbc2a216868c53f9245670a and then fail after it is applied?

tiopex, 2025-04-11:
Yes, both responders (umtp, cmtp) works before applying https://github.com/gregkh/linux/commit/3fc137386c4620305bbc2a216868c53f9245670a and fail after it is applied.

jfdelnero, 2026-11-01:
Seems to point to the 6.12.2 Kernel issue : 
https://github.com/viveris/uMTP-Responder/issues/130

closed 2026-11-01 by jfdelnero