viveris/uMTP-Responder #127: Android doesn't understand uMTP-Responder

viktor76525, 2025-03-21:
That is, Android (Samsung Tablet) isn't able to see / load any storage locations served by uMTP-Responder from a Raspberry Pi 4B.
A computer reads it fine but Android won't. Have also tried writing an Android app, using [MtpDevice](https://developer.android.com/reference/android/mtp/MtpDevice). When calling `MtpDevice.open()` it will return a successful result but only after roughly 10 seconds. Nothing actually works though as asking for storage IDs returns null. Other MTP devices like another Android device or GoPro work fine being read on Android.

Was using the latest release first but tried main just in case. Doesn't work.

jfdelnero, 2025-03-21:
- Which android version are you using ?
- Can you share some debug logs from uMTP Responder ?

> Was using the latest release first but tried main just in case. Doesn't work.

Do you mean this doesn't work the same way as the last release or differently ?

viktor76525, 2025-03-24:
Android 14.
The console output or something stored in /var/log ?
Doesn't work the same way.

jfdelnero, 2026-11-01:
I have tested the current version of uMTP-Responder (v1.10.1) on a Raspberry Pi Zero with these Android machines:

Samsung Galaxy J3  : Android 5.5
Samsung Galaxy A32 : Android 13
Samsung Galaxy Tab Active5 5G : Android 16

I managed to browse the uMTP disks and play some videos stored on the Raspberry Pi.

If nothing is detected on your end, this may be a USB cable/adapter issue. You need the right one to "tell" the Android machine to turn its USB port to Host/source mode. The issue may be related to the Rasperry Pi 4 or to the Kernel version used (the 6.12 was reported as non-working with any MTP responder: https://github.com/viveris/uMTP-Responder/issues/130 )

Please reopen if you have more details/logs to provide.

closed 2026-11-01 by jfdelnero